trezor/trezor-firmware #3892: Cannot authorize CJ on legacy fw - `Hard fault detected`

marekrjpolak, 2024-06-04:
**Describe the bug**
When trying to authorize CJ on legacy fw built from current `main` (both locally or in GH action), device shows `Hard fault detected. Please unplug the device.` On `1.12.1` or locally built firmware approximately from that time (March 2023) it works as expected.

I was testing it in desktop Suite with regtest and local coordinator.

onvej-sl, 2024-06-05:
I discovered that this bug
  * was introduced in a0aa414aa001a408bd67f58263b0ac3b6923f7e4,
  * doesn't only concern coinjoin, `tests/device_tests/bitcoin/test_multisig.py`, for example, causes the hard fault as well.

onvej-sl, 2024-06-07:
New discoveries:
  * The hard fault happens on [this](https://github.com/nanopb/nanopb/blob/c9124132a604047d0ef97a09c0e99cd9bed2c818/pb_decode.c#L1416) line.
  * It is caused by updating `gcc-arm-embedded`  from `10.3.1` to `12.2.1`.
  * Updating `gcc-arm-embedded` to `12.3.1` or to `13.2.1` didn't help.
  * Updating `nanopb` to `0.4.8` didn't help.

closed 2025-01-25 by mmilata

bosomt, 2025-01-27:
**QA OK**

[image]

**Info:**
 - Suite version: desktop 25.1.2 (2db40a5dfed7c8743da2ff2cce78e86e557a0aaa)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/25.1.2 Chrome/128.0.6613.186 Electron/32.2.6 Safari/537.36
 - OS: MacIntel
 - Screen: 1470x956
 - Device: Trezor T1B1 1.13.0 regular (revision 7bf7d5fac2e897aca792dece98d6346df5e29bd2)
 - Transport: BridgeTransport 2.0.33

PR #4535 (2025-01-25): fix(legacy): uint64 protobuf field decoding
Fixes #3892 hopefully.

It seems that some Cortex-M3 instructions support unaligned access and some don't? Couldn't find it explained in a single place, however it seems that STRD (64bit store) can't handle it.