trezor/trezor-firmware #3915: Device tests don't work against T3T1 hardware

mmilata, 2024-06-06:
Get stuck on the first tap-to-confirm screen. Reproducible on emulator by leaving animations enabled.

TychoVrahe, 2024-06-06:
the issue (or at least one of the issues) is that the layout does not change when swipe occurs, but later after some animation frames.

Thus, the notify_layout_change function is called too early and the test does not see the new layout.

Proper solution is probably introduce some notify layout change to EventCtx, call it whenever the layout really changes in rust and handle this flag from python by calling that notify_layout_change function

matejcik, 2024-06-06:
i'll see if GFL #3110 can help here too -- at least the framework is solid enough for it, but the fact that the flows now happen in Rust brings some challenges

PR #4375 (2024-12-03): Improve HW testing for T3T1
* Fixes #3915 by disabling animations for debug builds of firmware. This will also speed up HW tests on other models.
* Fixes the `DISABLE_OPTIGA` argument that got broken in the kernel/coreapp split
* introduces "storage insecure test mode" that cuts down the number of PIN PBKDF2 iterations, and does not erase unused storage banks when wiping. This shaves off ~2 seconds **per testcase**, which should save us ~1500 s on the overall run

animations are disabled on PYOPT=0 builds by default, but you can explicitly enable them by providing `TREZOR_DISABLE_ANIMATION=0`  [... cut]
closed 2024-12-03 by matejcik