trezor/trezor-firmware #3918: Missing end of sentence after dry run

bosomt, 2024-06-06:
**Describe the bug**
Missing end of sentence after dry run.

**Steps to reproduce:**
1. Run dry run 
2. Enter loaded seed
3. Success text is shortened by ...

**Info:**
 - Suite version: desktop 24.6.1 (ee1bf19e747daba31bec272b64d09bfa8c6a807e)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/24.6.1 Chrome/118.0.5993.159 Electron/27.3.8 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.7.1 regular (revision 83d1ce2b8e6e4eefe09539f06b28709c8a7a89dc)
 - Transport: BridgeTransport 2.0.33

**Screenshots:**
[image]

bosomt, 2024-07-23:
Guess its only valid for 20w backups

`    backup_type: Slip39_Single_Extendable (3),`

[image]

Hannsek, 2024-07-23:
The success screen was redesigned. It will be fixed in the upcoming release.

closed 2024-07-23 by Hannsek