trezor/trezor-firmware #3940: CI: finish migration to GitHub Actions

mmilata, 2024-06-11:
Still missing:

- [ ] trezor-user-env tasks
- [ ] ARM builds

Hannsek, 2024-06-12:
Jirka's tool.

Hannsek, 2024-10-30:
@mmilata isn't this done already?

mmilata, 2024-10-30:
Right, everything for trezor-user-env should be built in GitHub CI now. Not sure about the status of Jirka's tool?

There are scheduled pipelines that we may want to disable (aren't they currently eating company money? cc @vdovhanych): https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/pipeline_schedules

Can't log in to the private GitLab repo anymore, there may be some also.

vdovhanych, 2024-10-30:
I disabled the scheduled pipelines in GitLab, I don't think it was running though we disabled the billing account there some time ago. 

I belive that this is done basically.

Hannsek, 2024-10-30:
Jirka's tool is not done. But it is not that important now. 
So I am closing this issue. Thanks.

closed 2024-10-30 by Hannsek