trezor/trezor-firmware #3941: CI: speed up GitHub actions

mmilata, 2024-06-11:
Aside from the tests themselves, there are some things that will speed up everything:

1. cache nix environments
   * https://github.com/cachix/cachix-action
   * https://github.com/nix-community/cache-nix-action
   * https://github.com/DeterminateSystems/magic-nix-cache-action
2. cache poetry environments (or implement #2271 and 1.)
3. investigate why uploads to data.trezor.io are so slow
   * every filename is logged and there are lots, there's probably some option in awscli not to print them

cderinbogaz, 2026-09-07:
Disclosure: I work on [Runnerhut](https://runnerhut.com). For the Nix-cache idea and the proposed cargo-audit split, I would measure restored environment setup time separately from audit execution. A cache hit is only a win if downloading and unpacking it costs less than recreating it.

I also noticed the linked UI-fixture optimization, so I would use a post-#7205 upload baseline before treating uploads as an unresolved bottleneck. Faster CPUs would not by themselves establish an improvement there.

If compute remains a significant part after those changes, I can help outline a Runnerhut comparison for a compatible Linux x64 host job. Nix installation, disk needs and cache behavior would need checking first; I would not promise a speedup or savings over public-repository hosted compute.

mmilata, 2026-09-18:
Did some basic tests, the nix cache can decrease the environment setup time from about 3 minutes to 1 minute.

Another item is submodules. Checkout without submodules takes 2s, with submodules 30s. There are two separate issues:
* `actions/checkout` doesn't seem to support shallow clones of the submodules even if the main repo is shallow
* we can't run uv without `vendor/ts-tvl` and we can't run xtask without uv and `vendor/QR-Code-generator`

A solution would be to have own our action for checking out: call `actions/checkout` non-recursively, then do a shallow clone of the essential or all submodules, depending on a parameter.