trezor/trezor-firmware #3947: Receive address confirmed after xpub check

bosomt, 2024-06-13:
**Describe the bug**
Follow up of https://github.com/trezor/trezor-suite/issues/12775
Already fixed and during confirmation it always says XPUB.
But last success screen says receive address

**Steps to reproduce:**
1. Compare xpub via Trezor Suite
2. Check last screen after confirm

**Info:**
 - Suite version: desktop 24.6.1 (9d8fc586194255068350256208cbd828cc880876)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/24.6.1 Chrome/118.0.5993.159 Electron/27.3.8 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.7.2 regular (revision da75d8f4b67410b40a9cfd2954d183d81dd6e8e8)
 - Transport: BridgeTransport 2.0.33

**Screenshots:**

[image]

bosomt, 2024-10-21:
**QA OK**

**Info:**
 - Suite version: desktop 24.11.0 (17bac171ecd870c4d76bb64c6ec5f8d69b333d4c)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuiteDev/24.11.0 Chrome/128.0.6613.162 Electron/32.1.2 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.8.4 regular (revision da7ddd5c8c0b2040eef38247d4fd73d4576e6708)
 - Transport: BridgeTransport 2.0.33

obrusvit, 2024-12-02:
Unfrotunately, https://github.com/trezor/trezor-firmware/pull/4182 reintroduced the issue and It made it to the release branch `release/24.12` so I'm re-opening the issue. Sorry.
cc @bosomt @Hannsek

obrusvit, 2024-12-05:
Will be fixed in #4322

bosomt, 2025-02-20:
**QA OK**

`Public key confirmed` reported now 

**Info:**
 - Suite version: desktop 25.2.2 (dd7673ce898ea18b90daa91f21edbb25a4268c46)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/25.2.2 Chrome/128.0.6613.186 Electron/32.2.6 Safari/537.36
 - OS: MacIntel
 - Screen: 1470x956
 - Device: Trezor T3T1 2.8.7 regular (revision 8a254aa8eae82f99630df63f40e4d290066a3efc)
 - Transport: BridgeTransport 3.0.0-bundled.25.2.2

closed 2025-02-20 by obrusvit