trezor/trezor-firmware #3980: Touches are represented as double touch most of times

bosomt, 2024-06-26:
**Describe the bug**
Touches are represented as double touch most of times.
Same behavior when entering PIN or passphrase
Managed to replicate it on multiple devices and multiple times.

**Steps to reproduce:**
1. Setup device
2. Enter PIN code
3. Observe touch screen behavior

**Info:**
 - Suite version: desktop 24.6.2 (b319fb1a1e4e2fb85e398589dec064c26450bab0)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/24.6.2 Chrome/118.0.5993.159 Electron/27.3.8 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T2T1 2.8.0 regular (revision 3a039c505f48b97542e6118d9fb2dc867306770a)
 -  Device: Trezor T2T1 2.7.3 regular (revision a8f808822cf44ab4657d27451b5b7dd4b04d486c)
 - Transport: BridgeTransport 2.0.33

**Screenshots**

https://github.com/trezor/trezor-firmware/assets/31506317/341c9810-2c06-4a29-996f-1906d711a0e7

Hannsek, 2024-06-26:
Did you try multiple devices?

bosomt, 2024-06-26:
2x TT
1x TS5
same results

closed 2024-07-01 by matejcik

Perlover, 2024-07-07:
Same bug: v2.7.2

But what's worse - when you have to validate a payment by pressing and holding the green button - it starts flashing and the circular timer of the button hold is reset each time.

Hannsek, 2024-07-07:
It it fixed in 2.8.0 which will be released this week.