trezor/trezor-firmware #3981: Not connected button does not fit on homescreen

bosomt, 2024-06-26:
**Describe the bug**
Not connected button does not fit on homescreen, see video

**Steps to reproduce:**
1. Setup device
2. Lock it
3. Observe homescreen

**Info:**
 - Suite version: desktop 24.6.2 (b319fb1a1e4e2fb85e398589dec064c26450bab0)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/24.6.2 Chrome/118.0.5993.159 Electron/27.3.8 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: 
 - Transport: BridgeTransport 2.0.33

**Screenshots:**

https://github.com/trezor/trezor-suite/assets/31506317/1f1c696e-93c9-43f2-aeab-07aa1ca653bd

TychoVrahe, 2024-06-26:
I don't think so, i didn't know about it

closed 2024-07-01 by mmilata

bosomt, 2024-07-01:
**QA OK**

 - Device: Trezor T3T1 2.8.0 regular (revision dd4671a5104952ef505d28d1f9e94d1484b4607a)

PR #3983 (2024-07-01): fix lockscreen Not connected label hiding
fixes #3981