trezor/trezor-firmware #3988: build failures on ubuntu-latest?

prusnak, 2024-07-01:
ISTM that the current master does not build on ubuntu-latest in the github CI

See the following failures of the bitcoin-hwi project:

* https://github.com/bitcoin-core/HWI/actions/runs/9745167713/job/26892446343?pr=742#step:4:386

```
cc: error: unrecognized command-line option ‘-ftrivial-auto-var-init=zero’
```

* https://github.com/bitcoin-core/HWI/actions/runs/9745167713/job/26892446696?pr=742#step:4:403

```
embed/lib/assert.h:43:47: error: ‘__FILE_NAME__’ undeclared (first use in this function)
```

It would be great if these issues were fixed so the HWI CI works again.

matejcik, 2024-07-01:
both those problems are resolved by using gcc 12, which can be installed on ubuntu.
wondering if there is any way to make that clearer to users who run into these issues

prusnak, 2024-07-01:
> both those problems are resolved by using gcc 12, which can be installed on ubuntu.

Feel free to close the issue unless you want to keep it open to address this part:

> wondering if there is any way to make that clearer to users who run into these issues

I can't think of anything except for updating the build documentation to contain `CC=gcc-12` for Ubuntu.

matejcik, 2024-07-02:
i was thinking more along the lines of:
```
 #if __GNUC__ < 12
    #error "This code requires GCC 12.0 or later."
  #endif
```
which is the reason (along with docs update) to keep this issue open

prusnak, 2024-07-02:
> #error "This code requires GCC 12.0 or later."

Don't we also support Clang?

matejcik, 2024-07-02:
we do, i don't know the required version off the top of my head. 15 presumably?