trezor/trezor-firmware #4004: Build Error about 'stackdummy'

tarkanaktan, 2024-07-06:
I am getting error while doing the command 'make vendor build_boardloader build_bootloader build_firmware' in bootloader section.
The log can be seen below. 

../py/stackctrl.c: In function ‘mp_stack_ctrl_init’:
../py/stackctrl.c:32:32: error: storing the address of local variable ‘stack_dummy’ in ‘mp_state_ctx.thread.stack_top’ [-Werror=dangling-pointer=]
   32 |     MP_STATE_THREAD(stack_top) = (char *)&stack_dummy;
../py/stackctrl.c:31:18: note: ‘stack_dummy’ declared here
   31 |     volatile int stack_dummy;
      |                  ^~~~~~~~~~~
In file included from ../py/runtime.h:29,
                 from ../py/stackctrl.c:27:
../py/mpstate.h:323:23: note: ‘mp_state_ctx’ declared here
  323 | extern mp_state_ctx_t mp_state_ctx;
      |                       ^~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [../py/mkrules.mk:80: build/py/stackctrl.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/tarkan/Desktop/trezor/trezor-firmware/vendor/micropython/mpy-cross'
make: *** [Makefile:298: build_cross] Error 2

Can you help me with this?

prusnak, 2024-07-06:
Version of gcc?

tarkanaktan, 2024-07-06:
I have gcc 13.2.0 installed.

tarkanaktan, 2024-07-06:
That was my mistake. I havent done nix-shell inside of poetry shell. I made the build again, but I got this new error.

warning: `trezor_lib` (lib) generated 1 warning (run `cargo fix --lib -p trezor_lib` to apply 1 suggestion)
    Finished release [optimized + debuginfo] target(s) in 54.02s
scons: `build/bootloader/bootloader.bin' is up to date.
python ./translations/order.py
make: python: No such file or directory
make: *** [Makefile:226: translations] Error 127

It seems like a missing file. Am I missing some points?

tarkanaktan, 2024-07-06:
I added allias for python and sems ok. However now I started to get an below error.

arm-none-eabi-objcopy -O binary -j .header -j .flash -j .data -j .confidential build/bootloader/bootloader.elf build/bootloader/bootloader.bin
tools/headertool.py build/bootloader/bootloader.bin -D
Traceback (most recent call last):
  File "/home/tarkan/Desktop/trezor/trezor-firmware/core/tools/headertool.py", line 4, in <module>
    from trezorlib import cosi, firmware
ModuleNotFoundError: No module named 'trezorlib'
scons: *** [build/bootloader/bootloader.bin] Error 1
make: *** [Makefile:244: build_bootloader] Error 2

[nix-shell:/home/tarkan/Desktop/trezor/trezor-firmware/core]#

matejcik, 2024-07-08:
i vaguely remember that gcc 13 introduces some sort of detection that is hated by our version of micropython
that makes gcc 12 the only currently supported version.
nix-shell will give you gcc12

tarkanaktan, 2024-07-08:
Should I also run nix-shell inside of the poetry shell? I changed my gcc version to 12.3.0 and bootloader was built succesfully. As I understand the following logs.

   Finished `release` profile [optimized + debuginfo] target(s) in 0.05s
scons: `build/bootloader/bootloader.bin' is up to date.
python ./translations/order.py
No new items found.
python ./translations/cli.py gen
signatures.json is already up-to-date
./tools/build_templates

However, I am getting the same error for 'build_firmware'. You can find the log below. 
endor/micropython/py/stackctrl.c: In function 'mp_stack_ctrl_init':
vendor/micropython/py/stackctrl.c:32:32: error: storing the address of local variable 'stack_dummy' in 'mp_state_ctx.thread.stack_top' [-Werror=dangling-pointer=]
   32 |     MP_STATE_THREAD(stack_top) = (char *)&stack_dummy;
vendor/micropython/py/stackctrl.c:31:18: note: 'stack_dummy' declared here
   31 |     volatile int stack_dummy;
      |                  ^~~~~~~~~~~
In file included from vendor/micropython/py/runtime.h:29,
                 from vendor/micropython/py/stackctrl.c:27:
vendor/micropython/py/mpstate.h:323:23: note: 'mp_state_ctx' declared here
  323 | extern mp_state_ctx_t mp_state_ctx;
      |                       ^~~~~~~~~~~~
cc1: all warnings being treated as errors
Maybe it is because of the warnings being treated as errors. How can I solve this?

prusnak, 2024-07-08:
Are you sure you that gcc-12 is used for the build?

Maybe you need to use the following?

export CC=gcc-12

tarkanaktan, 2024-07-08:
It still gives the same error.  When I check my gcc version  I can see that it is gcc-12.
gcc --version
gcc (Ubuntu 12.3.0-17ubuntu1) 12.3.0

The funny part is I was having the same error in bootloader build. Now bootloader seems ok, but now the same error occurs in firmware build.
Maybe it is because the build option of -werror becuase it says that warnings will be considered as errors.

matejcik, 2024-07-09:
what's your cross-compiler version?
`arm-none-eabi-gcc --version`

i'm confused now because the `dangling-pointer` warning was introduced in GCC 12, I can locally get the warning to trigger in an example program, but the build passes fine for me

matejcik, 2024-07-09:
> Should I also run nix-shell inside of the poetry shell? I changed my gcc version to 12.3.0 and bootloader was built succesfully. As I understand the following logs.

> The funny part is I was having the same error in bootloader build. Now bootloader seems ok, but now the same error occurs in firmware build.

i think this was some sort of confusion, because bootloader build should _not_ invoke any part of micropython. is it possible that you ran `make build_boardloader build_bootloader build_firmware` and the error originated in `build_firmware` job? and then later you re-ran `build_bootloader` separately, avoiding the error?
if not, that would be concerning

matejcik, 2024-07-09:
I see the problem now. As part of `build_firmware`, a `build_cross` target is executed that will use your system gcc.
You can separately run it beforehand as `make build_cross CC=gcc-12` to avoid the problem.

Funnily enough, the `dangling-pointer` warning _was_ introduced in GCC 12 but does not trigger here. Honestly looks like a bug in that warning that was fixed in GCC13, breaking the build. I think we can hotfix it here

PR #4009 (2024-07-09): build(core): fix build for GCC13
we need to cherry-pick two commits into the micropython submodule, and adapt one of the fixes
matching: https://github.com/trezor/micropython/pull/12 (we'll need to merge that one manually by fast-forward so that the branch stays in sync with this submodule update)

fixes #4004
closed 2024-07-09 by matejcik