trezor/trezor-firmware #4005: Get rid of left swipe to the menu from the main flow

Hannsek, 2024-07-08:
https://satoshilabs.slack.com/archives/CL1D61PQF/p1720447387711949

Hannsek, 2024-08-02:
@TychoVrahe is this on you or anyone from the application team can do it?

obrusvit, 2024-08-02:
@Hannsek If I understand you correctly, you don't want the menu to be accessible by a swipe-to-the-left movement. This can be easily removed from all flows by me or anyone.

However, from the conversation it seems that the problem was interpreting a horizontal swipe as a vertical swipe which was improved here https://github.com/trezor/trezor-firmware/pull/4060 

Can you confirm you still want to remove the behaviour?

TychoVrahe, 2024-08-02:
Exactly. I would personally wait and see whether this still causes problems and only then go for a complete removal

Hannsek, 2024-08-02:
Alright, let's wait and see. 👍🏻

closed 2024-09-09 by Hannsek