trezor/trezor-firmware #4006: Incorrect More info screens during Multi-share Backup creation.

bosomt, 2024-07-08:
**Describe the bug**
Incorrect `More info` screens during Multi-share Backup creation from 20 word multishare ready backup

**Steps to reproduce:**
1. Navigate to `Settings/Device/Multi-share Backup`
2. Start generation after dry run
3. Check both `More info` screens

**Info:**
 - Suite version: desktop 24.7.1 (88dbe0410ac5b6245f392ce13e8c30f0f8f549f2)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/24.7.1 Chrome/118.0.5993.159 Electron/27.3.8 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.8.0 regular (revision dd4671a5104952ef505d28d1f9e94d1484b4607a)
 - Transport: BridgeTransport 3.0.0

**Screenshots:**

_number of shares_
[image]

_more info_
[image]

_treshold_
[image]

_more info_
[image]

Hannsek, 2024-07-08:
It is probably during the shamir recovery as well.

bosomt, 2024-07-08:
@Hannsek you are right
TT latest develop multishare creation 

[image]

 - Suite version: web 24.8.0 (814ab5d0df52939057bfa7643e7c0a83012e86f7)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:127.0) Gecko/20100101 Firefox/127.0
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.8.1 regular (revision e30a0e62313afd47df390846b3d33dc0bc2198f1)
 - Transport: BridgeTransport 2.0.33

closed 2024-07-22 by obrusvit