trezor/trezor-firmware #4019: Menu item text gets hidden if it is too long

matejcik, 2024-07-10:
**Describe the bug**
problem:
[image]

after a local hacky fix:
[image]

offending code: https://github.com/trezor/trezor-firmware/blob/700512901b822742759e677ae59fbaef04417842/core/embed/rust/src/ui/model_mercury/component/button.rs#L624-L635

**Why** do we even have this? ISTM we should default to showing both icon and text in all cases.

**To reproduce**

`trezorctl btc get-address -n m/44h/0h/0h/0/0 -d`
then click the menu

matejcik, 2024-07-10:
UI tests pass with the local hack, which just unconditionally runs the first branch in the above code. This tells me that the other branches are dead code.

obrusvit, 2024-07-16:
> Why do we even have this? ISTM we should default to showing both icon and text in all cases.

Taken from `model_t` UI.

ibz, 2024-09-18:
Indeed, not all menus are tested.

But what is worse? I don't see a case for a menu item with no text - a menu item with incomplete text is certainly better.

So, I see 3 issues here:

1) Is there a case against showing the text all the time? If not, maybe we should bite the bullet and just do it?
2) What should we do *if* we detect that text does not fit? (can we even detect that - easily - in code?) I would argue that - if possible - we should probably just somehow crash (only in testing / emulator?) - that would make the CI pipeline fail, and would make us shorten the strings. Probably a pain, but we would do it once and know that it is done.
3) Increasing test coverage is something we should be constantly striving for, but unrelated to this issue - so that should not hold us back from actually fixing this issue which is quite bad.

mmilata, 2024-09-19:
> do you know of a case where we want to render icon without text?

nope

> we can crash in debug mode, if detecting this is easy at run-time

Seems to be easy in this particular case since we already have the widths in variables? For general case we'd probably have to extend the text renderer. Or perhaps graphics driver to detect drawing off-screen, but that sounds like it will have its own problems.

ibz, 2024-09-19:
> Seems to be easy in this particular case since we already have the widths in variables? 

Right, it was a silly question from my side to start with. That's what the **if** in question does - detect when the text does not fit and hide it.

ibz, 2024-09-19:
On the other hand, I am thinking what is the gain of crashing if the menus are not walked-through by the tests / CI anyway.

There is a very slim chance somebody will manually get to that point while testing, and if they do, they will still notice the menu looks bad, no? Crashing at that point will just annoy somebody that is trying to do something unrelated.

So maybe the best way is to simply show the incomplete text?

Hannsek, 2024-09-19:
> So maybe the best way is to simply show the incomplete text?

Most probably, as on every other places.

matejcik, 2024-09-19:
> On the other hand, I am thinking what is the gain of crashing if the menus are not walked-through by the tests / CI anyway.

a lot of tests actually do involve menus

the point of this is to crash if a _translation_ causes a menu item to not render; we don't really check all languages screen by screen, so the crash will indicate that some translation string doesn't fit and we need to change it

ibz, 2024-09-19:
> a lot of tests actually do involve menus

Yes, but then shouldn't you have seen *some* UI tests diffs when you disabled the other branches?

matejcik, 2024-09-19:
> Yes, but then shouldn't you have seen _some_ UI tests diffs when you disabled the other branches?

i didn't run all the languages

ibz, 2024-09-19:
OK, I'll do that then and see what happens. The way it is now is very bad, anyway.

obrusvit, 2024-09-19:
Alternative is implementing multi-line texts. As was also desired by Figma designs. See e.g. 
[image]

I think the combination of approaches would be the best if feasible:
1. try to fit on one line
2. if fails, try to fit on two lines
3. if fails, crash in debug, show incomplete text in prod

Hannsek, 2024-09-24:
@lapohoda I think we can do that, right?

PR #4226 (2024-10-02): feat(core): allow two lines of text in menu items
Fixes #4019
closed 2024-10-02 by ibz