trezor/trezor-firmware #4046: Flashing Fail

tarkanaktan, 2024-07-20:
Hello,

I am trying to flash the built code to device. However when I do it I got an warning message that says "Warn : no flash bank found for address 0x00000000". This message will occur for all the binaries which are boardloader, bootloader and firmware. When I checked the bin files I saw all the binaries start from 0x0000000 address. However, when I checked the address "model_T2T1.h" file, I assume they should be started from different addresses.  I added the bin files' screenshot.  Could you help me with this?

[image]

TychoVrahe, 2024-07-20:
for T2T1, or Model T, the binaries should really be flashed to addresses in the file you mentioned:
BOARDLOADER to 0x08000000
BOOTLOADER to 0x08020000
FIRMWARE to 0x08040000 (first part) and 0x08120000 (second part)

you can maybe use the makefile commands `make flash_boardloader` etc to make it easier for you, if you are using compatible debugger for this 

the "addresses" in the files are not really addresses, just offsets inside the file, iow meaningless for this purpose

TychoVrahe, 2024-07-20:
in any case, i dont see a reason for calling any of this a bug so i am closing this issue

closed 2024-07-20 by TychoVrahe