trezor/trezor-firmware #4048: July 21, 2024 firmware 2.7.2 ver update partially disables hidden wallet account having 1 token with 2 accounts for token

F101lp, 2024-07-22:
July 21, 2024 firmware 2.7.2 ver update partially disables hidden wallet account having 1 token with 2 accounts for token. ( I had 2 separate BTC wallets under the 1 hidden wallet account .. no issues for over 1 year.. now only 1 is accessible.. 

**Firmware version and revision**
firmware 2.7.2 ( was 2.7.0 when all worked fine).

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: [e.g. Trezor Suite, 
 - OS: [e.g. Windows, PC and on laptop
 - Browser [e.g. Chrome,
 - Version [24.1.2 ... and on alternate laptop running 25.5.4 

**To Reproduce**
Steps to reproduce the behavior:  n/a .. updated the firmware from 2.7.0 to 2.7.2 on device was it.. cant go back ?
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
I expected BOTH my single hidden wallet accounts to be accessible.. rebuilding the wallet does noting .. unless i go back to earlier firmware version..which i may need 

**Screenshots**
If applicable, add screenshots to help explain your problem.
5 account s show but more than 8 exists on the hidden wallet
[image]
 

**Additional context**
Add any other context about the problem here.

Hannsek, 2024-07-22:
Hi, I am sorry but I didn't understand your issue. Can you please explain more?

Hannsek, 2024-07-23:
So, are you saying that after the fw update to 2.7.2 you cannot see second accounts for BTC and Matic? Is that right?

F101lp, 2024-08-04:
issue solved... unclear cause.. but all back no issue.. maybe clowdstrike issue involved not sure.. but TREZOR tokans are safe

closed 2024-08-04 by prusnak