trezor/trezor-firmware #4053: Screen sleep mode after lock-screen inactivity

userofreddit44, 2024-07-23:
**Context**: Using the TS5 as a FIDO/GPG hardware device and thereby having it plugged into USB 24/7 for convenience
**Feature**: I would prefer the screen to fall into some sleep mode after lock-screen inactivity with instant return after activity 
**Motive**: To ensure the longevity of the screen.

Maybe some-what related to issue https://github.com/trezor/trezor-firmware/issues/4036 but not really.

Hannsek, 2024-07-23:
Thank you for the request. We'll discuss that and will let you know.

matejcik, 2024-08-05:
> **Motive**: To ensure the longevity of the screen.

The screen on Trezor Safe 5 is a regular LCD screen with LED backlight, which won't really wear out before the rest of your device fails. So you don't need to worry too much.

That said, the lockscreen should be on lower brightness, the fact that it is not is probably a bug.

closed 2024-08-05 by Hannsek