trezor/trezor-firmware #4058: Reduce size of precomputed tables for ECC

onvej-sl, 2024-07-24:
# Introduction
## Curves
There are two elliptic curves used in firmware:
  * secp256k1 (the Bitcoin curve)
  * nist256p1 (also known as secp256r1)

We also use curve25519 and ed25519 but I won't address them in this issue.

## Operations
There are five elliptic curve operations exposed to the core firmware.
  * signature genration (`sign`)
  * signature verification (`verify`)
  * recovery of public key from signature (`verify_recover`)
  * derivation of public key from private key (`publickey`)
  * Diffie-Hellman operation (`multiply`)

## Implementations
There are two implementations of elliptic curve operations used by the firmware:
  * native (`tc_*` functions in `crypto/ecdsa.c`)
  * library (`vendor/secp256k1-zkp`)

The library implementation supports only `secp256k1`, whereas the native implementation supports both curves. I noticed that `multiply` is implemented always using the native implementation, even though the library implementation supports this operation.

## Precomputed tables

The native implementation can use a precomputed table to speed up elliptic curve operations. A precomputed table for every curve takes 36 KiB.

The library implementation uses two precomputed tables. The first one is 32 KiB in size and cannot be reduced. The second one is 8 KiB in size and can be reduced to 128 B at the cost of slower `verify` and `verify_recover`.  See [this](https://github.com/trezor/trezor-firmware/issues/1534#issuecomment-817688585) for more detailed information.

The table below shows the duration of the operations.

|                | native (36KiB table) | native (no table) | library (8KiB table) | library (128B table)  |
|----------------|----------------------|-------------------|----------------------|-----------------------|
| sign           | 27 ms                | 103 ms            | 17 ms                | 17 ms                 |
| verify         | 134 ms               | 210 ms            | 8 ms                 | 11 ms                 |
| verify_recover | 134 ms               | 210 ms            | 8 ms                 | 11 ms                 |
| publickey      | 23 ms                | 99 ms             | 16 ms                | 17 ms                 |

## Usage
As far as I know, [bip32 child key derivation](https://github.com/trezor/trezor-firmware/blob/main/crypto/bip32.c#L297) is the last place in firmware where the native implementation is used for `secp256k1`.
As far as I know, `nist256p1` is used only in fido, [slip-17](https://github.com/trezor/trezor-firmware/blob/main/core/src/apps/misc/get_ecdh_session_key.py) and [slip-13](https://github.com/trezor/trezor-firmware/blob/main/core/src/apps/misc/sign_identity.py).

# Suggestions
1. Use the library implementation for [bip32 child key derivation](https://github.com/trezor/trezor-firmware/blob/main/crypto/bip32.c#L297).
2. Use the library implementation for `mulitply` on `secp256k1`.
3. Remove precomputed tables used by the native implementation. There are two variants:
  a. Remove the tables for both `secp256k1` and `nist256p1.` This saves 72 KiB but makes fido operations slower.
  b. Remove only the table for `secp256k1`. This saves 36 KiB and doesn't affect the speed of operations, but it requires some refactoring.
4. Reduce precomputed table in the library implementation. This saves 8 KiB but makes verification a bit slower.

andrewkozlik, 2024-08-01:
Let's implement all 4 suggestions with variant 3 a.

Slowing down FIDO operations by presumably 80 ms is insignificant, considering all the other processing that needs to take place, and is worth the 36 kiB of flash. We can leave the tables in T1 if it doesn't incur extra work, since flash space is not a limiting factor there.

Point 4 slows down verification by 3 ms. This should only be relevant for transactions with a large number of external inputs, and even in those cases it shouldn't add up to a large slowdown. Probably worth the 8 kiB. Again, not needed on T1.

Hannsek, 2024-11-06:
@onvej-sl is this done? Can we cloe it?

onvej-sl, 2024-11-06:
Done in https://github.com/trezor/trezor-firmware/pull/4083.

closed 2024-11-06 by onvej-sl