trezor/trezor-firmware #4080: Unable to reproduce T3T1 firmware hash

wizeguyy, 2024-08-01:
I am unable to get matching `sha256sum` for the published release of 2.8.0

As described in https://github.com/trezor/trezor-firmware/issues/3663, I use dd to overwrite the signature in the t3t1 firmware header:
```
dd if=/dev/zero of=trezor-t3t1-2.8.0.bin bs=1 seek=1471 count=65 conv=notrun
```

The hash I get is:
```
b8583a40f5081b70b7b9db089cec903d73610aefa9ac24d7aa49fda6090054af  trezor-t3t1-2.8.0.bin
```

When I checkout `core/v2.8.0` and run `sudo bash build-docker.sh core/v2.8.0` it completes with the following fingerprints:
```
Docker image retained as trezor-firmware-env.nix__core_v2_8_0
To remove it, run:
  docker rmi trezor-firmware-env.nix__core_v2_8_0

Built from commit a3a7b0cd7a8e0fe78726ddb1c3159c2972a7278a

Fingerprints:
5673e8cfc0f6cbae7a5e6c5b26473acdbda36a81d0fb074a680f4fba734690e1 build/core-R/bootloader/bootloader.bin
cf3ce230a69a681199f74cf6ac8c6c431f8fa7e0d0183437f93c5cc029fbd155 build/core-R/firmware/firmware.bin
5673e8cfc0f6cbae7a5e6c5b26473acdbda36a81d0fb074a680f4fba734690e1 build/core-R-bitcoinonly/bootloader/bootloader.bin
ae088439d44fc8643b8de28e0d7a8720cd3dbb247619f2742604bbe884542558 build/core-R-bitcoinonly/firmware/firmware.bin
85554b70fd79f58495052ea4e82298495b0a199d395efb072519b56ee4affad7 build/core-T/bootloader/bootloader.bin
d6b9fb54029757a289aebd9eb65c474a68fbe808846f703dc37372105e148674 build/core-T/firmware/firmware.bin
85554b70fd79f58495052ea4e82298495b0a199d395efb072519b56ee4affad7 build/core-T-bitcoinonly/bootloader/bootloader.bin
27f5376acc797d1b0f36f61a47bb61f6a91cfd03ac5a7a09d5e5d98633956a4f build/core-T-bitcoinonly/firmware/firmware.bin
e3e82a554c1904d37f2183ad85061c3bacb54c7cc82e8b87938119b55f9f190f build/core-T3T1/bootloader/bootloader.bin
bd199ce0934769aca5c3a91f71fd48e533d88c8cf087b76ac49db415fa08c286 build/core-T3T1/firmware/firmware.bin
e3e82a554c1904d37f2183ad85061c3bacb54c7cc82e8b87938119b55f9f190f build/core-T3T1-bitcoinonly/bootloader/bootloader.bin
3dc847cc396fe83f5a324242097a4cf97fc64acf90516efcfcf23b6d3103a992 build/core-T3T1-bitcoinonly/firmware/firmware.bin

```

I then manually perform the hash as described in the docs:
```
$ sha256sum build/core-T3T1/firmware/firmware.bin
26f136b85f553e3f085075733e3548e088b58ffb4279b5424fd355ba4d4efc92  build/core-T3T1/firmware/firmware.bin
```

Still, none of these match the hash I have for the published release. Did the firmware headers change? Am I missing something?

prusnak, 2024-08-01:
What host OS do you use?

matejcik, 2024-08-01:
T3T1 vendor headers are 1024 bytes so you'll need to update the `dd` offset.

We really should document this...

closed 2024-08-01 by matejcik