trezor/trezor-firmware #4084: Remove 50 character limit from passphrase

gernotpokorny, 2024-08-04:
I cannot write in #203, because it got archived. I hope you don't mind me writing this here. If you don't like it, then please inform me and I open a new issue.

I do not agree at all that this is no important issue.

Please let me explain.

Actually the 50 character limit is a major issue in regards to wallet recovery in case of a single letter failure within the backup (or much worse multiple letters, which I explain below). 

If you use a long passphrase, then using a 12 word bip39 seed phrase as a passphrase with a checksum actually makes sense from a wallet recovery perspective in regards to storing your crypto safe and securely if you want to use a passphrase. I read on reddit that many people are doing this it seems with other wallets, but it is not possible with Trezor. Having a checksum in the passphrase is useful for recovery purposes in case of errors within the backup. If you don't use a bip39 seed as a passphrase, but instead a 50 letter random passphrase, you essentially made your chance of wallet recovery much worse in case for example two characters are missing within the backup. Even if you are using as an example a 20 character random string as a passphrase it is also much more susceptible to errors within the backup. You do not want your passphrase become the weak point, but with the 50 character limit it becomes the weak point.

With a bip39 seed phrase as a passphrase you can use common recovery tools on an offline PC in case of errors within the backup. The checksum is very useful there. If you do not have a checksum and you miss two letters of a 20 character random string passphrase then I don't see how you would be able to recover from that without spending an immense amount of time writing code and exposing the seed on a PC connected to the internet trying to brute force and check if funds are in the hidden wallet. And if too many characters are missing that becomes unfeasible. That sounds absolutely horrible to me and a situation that nobody wants to be in.

Actually this is one thing that I really do not like about Trezor, because this is bad and there is no argument to not remove this limit. The argument made by the Trezor team was it is not important enough, but as I outlined above it is definitely important and can potentially result in the loss of funds.

By the way using only the first 4 chars of a 12 word bip39 seed also weakens the ability to recover a wallet compared to using the full words, if you would want to suggest that. It would also become the weak point if your actual recovery seed backup does use full words, which it always should with a 12 word seed.

50 random characters without checksum is much more error prone to backup failure then a longer passphrase comprised of a 12 word bip39 seed which includes a checksum. 

The 50 character limit for the passphrase decreases the ability for people to secure their crypto securely if they use a passphrase as outlined above.

This is my understanding of this issue.

I would be happy to hear that this 50 character limit finally gets removed for the good of all.

Thank you for taking it into consideration.

prusnak, 2024-08-04:
I don't think there is a passphrase length limit for the core models (every mode except Model One). Or is it @matejcik?

gernotpokorny, 2024-08-04:
Sorry I forgot to mention I am talking about Trezor Safe 5.

matejcik, 2024-08-05:
> I don't think there is a passphrase length limit for the core models (every mode except Model One). Or is it @matejcik?

There is a 50 char limit on all models -- this was originally intended for compatibility with T1, so that you can always restore your Trezor T wallet on a Trezor One. I suppose we can revisit this decision now that we have TS3 as the entry-level model in the lineup?

prusnak, 2024-08-05:
> There is a 50 char limit on all models -- this was originally intended for compatibility with T1, so that you can always restore your Trezor T wallet on a Trezor One. I suppose we can revisit this decision now that we have TS3 as the entry-level model in the lineup?

I agree that we should revisit this. I would be fond of getting rid of the limit for core models and recommend people who want longer passphrase on T1 to upgrade to newer models.

matejcik, 2024-08-05:
with that said, @gernotpokorny, yours is not a good argument!

> Having a checksum in the passphrase is useful for recovery purposes in case of errors within the backup. If you don't use a bip39 seed as a passphrase, but instead a 50 letter random passphrase, you [...]
(...)
> With a bip39 seed phrase as a passphrase you can use common recovery tools on an offline PC in case of errors within the backup.

If you make your passphrase a random Bitcoin address, your passphrase will be 42 characters with a significantly stronger checksum than that of a bip39 seed phrase, plus the option of error recovery. (with up to 160 bits of entropy, depending on how exactly you generated the "random" address -- if it is generated from a random 12-word seed, it's going to have 128 bits)

If you are using a password manager to store your passphrase, you don't care about these things because you will be copy-pasting the passphrase anyway.
If you are re-typing it from paper into a PC, then just skip the middleman and store it in a password manager. Chances are it will be more secure that way.
And if you are re-typing it from paper into a Trezor device directly, then a much more limiting factor is going to be _you_ making the typos. It does not matter if there is a checksum or not, if you don't manage to actually enter the right characters. (and let's be honest here, Trezor devices aren't really suited for typing long texts.)

> By the way using only the first 4 chars of a 12 word bip39 seed also weakens the ability to recover a wallet compared to using the full words, if you would want to suggest that. It would also become [...]

First 4 chars of each word _uniquely determine_ the full word. That means that using first 4 chars is fully equivalent to using the full words. If your recovery tool can't deal, you can simply go through the list by hand and for each 4-char sequence, find _the only_ word that starts with that sequence.

> if your actual recovery seed backup does use full words, which it always should with a 12 word seed.

Many, perhaps most, backup products work by only storing the first four characters -- which, as I said above, is _fully equivalent_ to storing the full words.
Most wallets will also auto-complete the full word from the first four characters.

gernotpokorny, 2024-08-05:
> I agree that we should revisit this. I would be fond of getting rid of the limit for core models and recommend people who want longer passphrase on T1 to upgrade to newer models.

Sounds great.

> with that said, @gernotpokorny, yours is not a good argument!

I do not agree. I think you misunderstand the point of the checksum that I tried to make. See below.

> If you make your passphrase a random Bitcoin address, your passphrase will be 42 characters with a significantly stronger checksum than that of a bip39 seed phrase, plus the option of error [...]

But you cannot recover it without a program if a few characters are not readable. Small errors in your backup (for example a few characters of your passphrase are destroyed on your paper backup or steel backup) you can correct by yourself just by looking at the wordlist. You cannot correct things yourself easily if two characters are missing in a Bitcoin Address, it is a more difficult task.

> And if you are re-typing it from paper into a Trezor device directly, then a much more limiting factor is going to be you making the typos. It does not matter if there is a checksum or not, if you [...]

I think you misunderstand the point about the checksum. The checksum is not there to ensure you type the right passphrase. I know that the only way to see if I typed it correctly is to see if my wallet loads the correct balances and addresses after I typed it. The checksum just helps to recover your passphrase in case parts of your backup are destroyed. For example if you use some type of form of steel backup and due to some event a few characters are not readable anymore, then you have a huge advantage in regards to recovery of your passphrase if you use a BIP39 seed phrase as a passphrase compared to a Bitcoin private key or some other random string.

> First 4 chars of each word uniquely determine the full word. That means that using first 4 chars is fully equivalent to using the full words. If your recovery tool can't deal, you can simply go through the list by hand and for each 4-char sequence, find the only word that starts with that sequence.

Of course the backup tool like a steel backup can deal with 4 chars, but it is not best practice to use 4 chars for a 12 word seed if the steel backup for example supports both, because using the full words gives you a higher chance of recovery in most cases if the backup gets damaged and characters are lost.

Note I do not say 4 char steel backup constructions are bad, this is not what I am saying. 4 char steel backup constructions make sense in come cases, because the constructions gets more durable if it is smaller. But nevertheless there are many steel backup solutions which have the same size for 12 full words and using 4 chars and there it makes sense to use the full words.

> Many, perhaps most, backup products work by only storing the first four characters -- which, as I said above, is fully equivalent to storing the full words.
> Most wallets will also auto-complete the full word from the first four characters.

I don't know if most backup solutions use 4 chars. There are so many with full words.

If my backup supports full words and 4 chars, then I probably would use full words for the reason of better chance of recovery in case the backup gets damaged. At lease this is also what the manufactures suggest to do if the steel backup supports both.

Furthermore just as a small sidenote, it is much easier to see that your passphrase is actually comprised of bip39 seed words, if you use full words in case you do not remember that fact that you used a a bip39 seed for the passphrase. Knowing that gives you a better chance of recovery of the passphrase in case of damage. 

I also want to emphasize that the passphrase becomes in a way part of your secret, you cannot access your funds without it. That means you have to ensure it is not weaker then your seed in regards to damage resistance and so on. What I want to avoid is that the passphrase becomes the weak point of the secret.

avguima, 2026-08-27:
Hi @kanakx we are working on an update, it should be available in the next couple of months.

kanakx, 2026-08-27:
Hello @avguima, thank you so much for your response! I can't express how happy I am to hear that you're working on it!!! 🥹

closed 2026-09-02 by romanz