trezor/trezor-firmware #4097: Memonic Keyboard does not suggest SLIP39 Word

shimunn, 2024-08-09:
**Describe the bug**
A clear and concise description of what the bug is.

I'm trying to recover my Safe 4 using a 20 Word samir share, while entering one of those 20 word an totally different word is suggested preventing me from recovering.

**Firmware version and revision**

2.8.1

**Desktop/smartphone setup (please complete the following information):**

Trezor Suite

**To Reproduce**

Recovery using 20 word samir-share

**Expected behavior**

The correct word is being suggested

**Additional context**
Checked my seed, as well as SLIP39 the desired word should be in the dictionary. I've also tried to recover using the wrongly suggested word hoping for it to just be an display error.

prusnak, 2024-08-09:
Just click once, not multiple times. 

Eg, if you want to enter "B", do not push the button twice, just once.

Hannsek, 2024-08-09:
The keyboard is T9. Here, every word from slip39 checklist has unique sequence of 4 keyboard buttons in defined order.

shimunn, 2024-08-09:
I know how the keyboard works, all other words enter just fine. That specific word however does not match the suggested word at all, all characters are different

prusnak, 2024-08-09:
> I know how the keyboard works, all other words enter just fine. That specific word however does not match the suggested word at all, all characters are different

How many times did you press button for each letter?

shimunn, 2024-08-09:
Until the desired character appeared. Apologies for my unclear device description(I'm not used to using my laptop keyboard) I'm using the Safe 5. The first character required just once click.

prusnak, 2024-08-09:
> Until the desired character appeared. Apologies for my unclear device description(I'm not used to using my laptop keyboard) I'm using the Safe 5. The first character required just one click.

Each character requires only one click. Even when they are not the first on the button.

matejcik, 2024-08-10:
That does not matter though, because there are no SLIP39 words where you even _can_ press the same button twice.
Hence the same button becomes essentially unresponsive after the first click.

@shimunn to be extra clear:

when inputting your word

You press, in sequence, the four buttons that match the first four letters of the word.
[image]

For instance, to enter the word `desire`, first four letters are `desi`, you press:
1. `cd` (once, or any number of times you like)
2. `ef` (dtto)
3. `rs`
4. `ghij`
then the word `desire` will appear as a suggestion and you can confirm it.

It does not matter if there is a _different_ suggestion throughout the process; continue clicking the buttons.

Are you saying that, if you follow this procedure, the word still cannot be entered?

shimunn, 2024-08-10:
Finally got to the bottom of this, I've entered the last two character in the wrong order after all. Thanks for the for all the helpful and swift input. I still think it's quite counterintuitive to suggest an word starting with an different prefix if one has gone through the effort to enter the exact characters.

closed 2024-08-10 by prusnak