trezor/trezor-firmware #4102: Touch Screen Panel wasn't loaded properly Fault

OD111, 2024-08-13:
I flashed STM32F429 Discovery board with Trezor T model firmware and I connected sd card reader and touch screen with jumper cables like in the photo.
 I got Touch Screen Panel wasn't loaded properly, then I visited trezor.io/rsod. They said "Try rebooting the device." I tried then I fed the screen a different power supply but the fault didn't change.

I used this page(https://github.com/trezor/trezor-hardware/blob/master/electronics/trezor_model_t/index.md
) to set up the system. 

I already haven't disassembled TFT LCD display and SDRAM chip. Could the un-disassembled components be the root cause of the problem?

[**Firmware version and revision** 
I used this path to create binary files.
(https://github.com/trezor/trezor-firmware/blob/main/docs/core/build/index.md)

Fault Photo
[image]

prusnak, 2024-08-13:
> I already haven't disassembled TFT LCD display and SDRAM chip. Could the un-disassembled components be the root cause of the problem?

certainly

closed 2024-08-13 by prusnak

OD111, 2024-08-13:
Thank you, I will share results.