trezor/trezor-firmware #4192: [Feature Request] - Implement a Thief's PIN

ffrediani, 2024-09-18:
I came to find out more about this feature and found there is already a Wipe PIN feature (#63 and #720) which may be useful for certain scenarios, but not for all. One of them is when someone takes possession of your Trezor and forces you to reveal your PIN.

If you give the person the Wipe PIN the person will be able to see the device was wiped out and may threat you further and escalate the situation.

If there is a thief's PIN you give this one which will unlock Trezor normally but another wallet with little or not funds.

I understand the Hidden Wallet feature and although similar that is not quiet the same thing as it requires a passphrase which may not be desirable to be used for different reasons.

Could you consider implementing a Thief's PIN and an extra option for such scenarios ?

hynek-jina, 2024-09-18:
What do you miss with passphrase scenario?
- You reveal PIN to your default account.. and have most of the bitcoin behind unknown passphrase 🤷

ffrediani, 2024-09-18:
There are scenarios where the passphrase use is not desirable. Depending on your seed backup strategy it may not be feasible those keeping the backup have to remember of a passphrase as well. 
Having an specific PIN just for this propose simplify things a lot and may help to dismiss the situation and avoid escalating it.

Hannsek, 2024-09-19:
> There are scenarios where the passphrase use is not desirable.

Can you please elaborate on this?

> Depending on your seed backup strategy it may not be feasible those keeping the backup have to remember of a passphrase as well.

You can set the passphrase to a set of number, just like your PIN. 🤷🏻‍♂️

prusnak, 2024-09-19:
It's not worth complicating the design for user story that can easily be done with a decoy passphrase.

ffrediani, 2024-09-19:
@Hannsek I think you didn't get my point. Let me elaborate further.
@prusnak there is extra complication. There is a solution for a problem that there is no better solution for a particular and important scenario.

Wipe PIN feature is good and may be used on certain situations where the owner has possession of his Trezor. When used it shows a confirmation on the screen the device was wiped.

In a critical situation like a if a thief takes possession of your Trezor and threats you to revel your PIN either if you type it or give it to the person they will be able the device was wiped and escalate the situation further more which is undesirable.

The problem with passphrase is that it may not be desirable to be used in certain scenarios and doesn't necessarily have to do with a a threatening situation. When you have you seed held by trustworthy people, there people may not be very technical or know how to use if necessary. An extra passphrase is a complication and people may feel secure enough for backup proposed just to teach them how to restore using the seed words. These trustworthy people don't have your PIN and don't need it also.

The request here is just to add a option to setup a Thief's PIN inside Danger area for this type of scenarios which may be real to people so they have an extra chance to protect their funds. There is no need to complicate or develop anything complex. The same way there is the Wipe PIN add an extra configuration to add this other PIN and when typed it should unlock the Trezor normally for who is looking at it, but open other wallet instead where you may deposit little crypto to have a chance to distract the thief.

Hannsek, 2024-09-20:
> to add this other PIN and when typed it should unlock the Trezor normally for who is looking at it, but open other wallet instead where you may deposit little crypto to have a chance to distract the thief.

Exactly this behaviour is possible by using decoy passphrase.

matejcik, 2024-09-20:
> In a critical situation like a if a thief takes possession of your Trezor and threats you to revel your PIN either if you type it or give it to the person they will be able the device was wiped and escalate the situation further more which is undesirable.

If the thief goes as far as physically threatening you, they may not be happy with the amount of funds you left on the thiefs PIN and escalate anyway. (this has happened in the real world)

In a high-stress situation, don't try to be clever. **Don't put your savings above your life.**

> There is no need to complicate or develop anything complex.

ha ha.

One, the code that deals with PIN and wipe code is one of the most complex and sensitive parts of Trezor. It must be, so that it is difficult for attackers to break the PIN.

Two, Trezor is designed to hold exactly one seed, available after decrypting it with your legitimate PIN. It's highly unclear how we'd stuff a second, decoy wallet in there, in a manner that allows you to store funds and retrieve them later, how this would interplay with device recovery, etc.

Your idea sounds good on the surface, but it is costly in terms of R&D time, adds unnecessary complexity to a highly sensitive place in the overall security model of the device, is only _possibly_ useful in an extremely specific situation (which you should try to avoid in the first place), and you can cover the same usecase by using a passphrase.

closed 2024-09-20 by matejcik

ffrediani, 2024-09-20:
It seems you focused on finding reasons not to do than to understand the scenarios. Let's try to split each point separately in order to dismiss them. You are analyzing the happiness of the thief about the funds found and not the possibilities of users (your customers) may have in such odd situations. And you to justify your point you put to give your funds, funds protect by a hardware wallet. Unbelievable !

Back to more technical stuff, this feature request has not to do with the Wipe PIN. It was mentioned just as an similar example. As it is already implemented, if there is something that can be re-used from it there will not be a lot more work and research, if it is a different implementation than it is another analysis. Wipe does something, unlocking (a different wallet) does something else.

The point about holding only one seed is a valid point to be discussed. I don't have that answer or suggestion right now and it needs thought, but there may be other alternatives to be considered to reach such result.

You didn't put enough information to show and examples on how so complex this is in terms of R&D. If there are other ways already implemented like the normal PIN, Wipe PIN, use of passphrase what else would be so complex and new that would be so costly in terms of R&D. Doesn't sound a complete write from scratch.

You end up by trying to blame the user for being assaulted by a thief as it is something he could always avoid. Perhaps you confuse the possibility to split a seed backup (for example with Shamir backup) then with something that has a single point of failure which is someone forcing you to type a PIN and transfer funds from your hardware wallet.

Passphrase is not an option for all scenarios and it was explained the reasons already. Can we get over this ?

prusnak, 2024-09-20:
> Can we get over this ?

Please respect our decision why not to implement this feature. Thank you!

ffrediani, 2024-09-20:
@prusnak is it already decided ? By who ? Do you have authority to decide this on your own or the person who closed the issue  and do you talk on behalf of the company or the product team ? Or is it just your personal view that doesn't match, so you dismiss because you are not interested without knowing if this reflect the rest of the people's view ? It is easy to close other's issue and not to provide proper information.

I explained and gave examples where an existing feature may not fit and this is real. What else would you expect ? I can elaborate more but I need to understand what you didn't understand if you are interested in put more details.

ffrediani, 2024-09-20:
@prusnak is it already decided ? Did you make that decision yourself with as little as this discussion ? Or is it just your personal view that doesn't match ?

I explained and gave examples where an existing feature may not fit and this is real. What else would you expect ? I can elaborate more but I need to understand what you didn't understand if you are interested in put more details. Although passphrase was implemented and can be used in certain scenarios, it can't on this particular one which is different. That is just that.

prusnak, 2024-09-20:
> Did you make that decision yourself with as little as this discussion ?

The issue was closed 10 hours ago by main maintainer of this repository, matejcik. Nothing to do with my personal view.

> it can't on this particular one which is different.

Please describe your scenario and why passphrase cannot be used in that scenario.

ffrediani, 2024-09-20:
Ok @prusnak thanks for the interest.

The point about passphrase is about seed backup.
There are several different ways to store your seed, from a simple paper, encrypted in a file, engraved in a metal to as more secure as a shamir backup. But there are others factors to be taken into account as the trustworthy people you entrust to keep them for you and as you may know in most people's life there aren't many. From these people some of all of them may not be much technical and they still need to follow instructions to access the funds in your absence.

In certain scenarios a passphrase adds complexity for them and may not be suitable to ensure they will be able to access the funds should it be needed.

This feature request is to give the user an extra possibility to a particular scenario to distract an agressor.
The Wipe PIN is a good option if the user can access the wallet **before** being threaten.

Should you still think that is undoable, perhaps an alternative way to look at this is to add an option to the Wipe PIN feature to choose not show one the Trezor screen that device was wiped upon entering the Wipe PIN.

prusnak, 2024-09-20:
Ok, now I understand your use case. 

We discussed the motivations of an attacker for very long time when designing the wipe PIN and I still do believe that a passphrase or wipe PIN or combination of these two is enough for your use case. I understand you disagree and see things in a different light, but that itself is not a good reason why to implement this feature.

ffrediani, 2024-09-20:
Ok prusnak thanks for getting back.
I wished you implement this not just because I agree or disagree with your point of view of when you designed these other features, but to give me and other customers further possibilities to deal with such scenarios and secure our funds. I understand there are countless possibilities and it is impossible to implement one for each, but some for most cases.
As mentioned Wipe PIN may be a solution depending on the scenario as you cannot predict. Passphrase is not and I doubt this will be for many others, specially more basic users.

Well, I guess you may not be willing to consider even possible improvements on the Wipe PIN feature as suggested above. If not thanks for the discussion and for sharing your point of view anyway. Appreciate the time and attention.

ffrediani, 2024-11-13:
@Hannsek @prusnak instead of what has been proposed above, which apparently would require more complex changes in the firmware, what about doing something simpler and improve the existing Wipe PIN feature to allow users with a toggle on/off option in order to show or not in the Trezor screen a message confirming the wipe.
If the user, in Trezor Suite toggles that off, when entering the Wipe PIN it would immediately wipe the device without any warning or confirmation.

That would be useful in a stress situation where you are forced to enter or give out your PIN. The attacker wouldn't be able to easily understand what happened and remove any possibility of immediate theft of resources.

ffrediani, 2024-11-18:
@Hannsek @prusnak did you have a chance to check this ?

ffrediani, 2024-11-18:
@prusnak thanks for getting back. Seems you don't appreciate much challenging or different point of views, including from customers. Seems what you discuss internally is enough to have things ever reconsidered or discussed further (after all it was years ago). You said you fail to see and you don't seem interested in making any effort to be convinced by a different person from your surroundings. Seems what has been done (which I am not criticising - I like the product very much), is definitive. It is not nice this more closed behaviour. Partially I understand it from more technical people. I am not demanding something to be done, rather just to be heard for a situation that is real, to discuss a different point of view, even if it is different from your entire team.

I heard you when you said the original request was more complex and would require complex firmware changes and changed 
the request to something much simpler and that would require minor changes, but still seems nobody is minimally interested.

Regarding what you fail to see, maybe it will keep going to /dev/null but it is simple to explain: The Wipe PIN feature can continue working as originally designed for those who prefer the confirmation (and can still be the default option). What has been asked is to give people the option to toggle on/off to show the confirmation in Trezor's screen.

In a situation where you are forced to reveal your PIN, giving the Wipe PIN or typing, if it doesn't show the confirmation the device is being wiped it may confuse the attacker that may not be used with it and understand what happened, so it allows to save time leaving only the seed phrase as a way to withdraw funds which normally is not readily available with the Trezor.
People may carry a Trezor used to do day to day transactions and the chances to be in such situation are higher. In these situations being able to wipe the Trezor after you have been challenged and having an extra chance to confuse the attacker is something useful to these scenarios.

Regards

matejcik, 2024-11-19:
@ffrediani,

> I am not demanding something to be done, rather just to be heard for a situation that is real, to discuss a different point of view, even if it is different from your entire team.

When @prusnak says "we discussed this years ago", what he really means is "we discussed _the argument that you just presented_ years ago".
The point here is you are not bringing any new arguments or changed situation to the table, besides "but _I_ would like it that way". It is our stance that implementing this option does not bring enough benefit to offset the development, maintenance, and UX costs -- and that calculation _includes_ the fact that your second proposal is relatively easier(*) to implement, _and_ that there are real people who would like to use this feature.

(* not _easy_, still, because we're still talking about the most sensitive and complex piece of code in the whole Trezor codebase)

Plus, I would like to stress again that **you should not put your savings above your life** and try to trick a thief who wields potentially deadly force. Because that's how, in real life, actual people actually die. That is _also_ a term that figures in the above value-proposition calculation.

ffrediani, 2024-11-19:
OK @matejcik thanks for your input so far.
As mentioned, I am not demanding in any way, but to have this discussion. As an intense user and someone who enjoys the product there are scenarios that may be seen different. I reckon the necessary technical work to change something well established, but in the other hand as usages change from time to time, see stuff that can be improved.