trezor/trezor-firmware #4217: Slow loading of homescreen

bosomt, 2024-09-24:
**Describe the bug**

Slow loading of homescreen or delay before showing homescreen
https://satoshilabs.slack.com/archives/C054XGU77QV/p1727184779370469

**Steps to reproduce:**
1. a
2. b
3. c

**Info:**
 - Suite version: desktop 24.10.0 (55ac8cb8a4edc1ac546c163acb3482cf53c4e4f8)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuiteDev/24.10.0 Chrome/124.0.6367.243 Electron/30.3.1 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T2T1 2.8.4 regular (revision ab96ce7954c7db384b26bc863d8752150027adbf)
 - Transport: BridgeTransport 2.0.33

**Screenshots:**

https://github.com/user-attachments/assets/6c49a419-f775-4779-a759-bdbcb0e15c4a

bosomt, 2024-09-24:
Im not able to reproduce it on TS5

**Info:**
 - Suite version: desktop 24.10.0 (55ac8cb8a4edc1ac546c163acb3482cf53c4e4f8)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuiteDev/24.10.0 Chrome/124.0.6367.243 Electron/30.3.1 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.8.4 regular (revision ab96ce7954c7db384b26bc863d8752150027adbf)
 - Transport: BridgeTransport 2.0.33

TychoVrahe, 2024-09-25:
closed via #4218

closed 2024-09-25 by TychoVrahe

bosomt, 2024-09-25:
**QA OK**

**Info:**
 - Suite version: desktop 24.10.0 (acd58e6c26394eb09907791dd4006d03c130fe65)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuiteDev/24.10.0 Chrome/124.0.6367.243 Electron/30.3.1 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T2T1 2.8.4 regular (revision 454b8140ecb1575a1adafb8e57051e345a32255e)
 - Transport: BridgeTransport 2.0.33