trezor/trezor-firmware #4230: Sui Support

poelzi, 2024-09-30:
Sui is a 3rd generation chain (DAG) with real world 700ms finality and 300 ms owned object finality with nearly infinite tps and dozens of other innovations. Fastest growing chain, easiest to program.

Not supporting Sui will exclude the next wave of web2 users.

**Design:** [Figma](TODO)

**Spec:** [Notion](TODO)

https://sui.io/

poelzi, 2024-09-30:
For me personally, trazor became useless since Sui Wallet does not support trazor and I moved most of my assets there. All other chains became uninteresting for me.

prusnak, 2024-09-30:
Not planned

closed 2024-09-30 by prusnak

poelzi, 2024-10-01:
Since you asked, this ticket is my reason for 2 stars

https://www.trustpilot.com/reviews/66facec45798237cc1c19a60

Sui will revolutionize the internet, and if you don't understand it, you have not understood the tech.