trezor/trezor-firmware #4254: Reproducibility of Trezor Safe 5 firmware v2.8.3

xrviv, 2024-10-11:
Greetings team Trezor! 😊 
Danny here from walletscrutiny 
---
**Describe the bug**
1. Hashes do not match for built firmware vs official firmware
  - We would like to request for the correct values for signature zeroing
  - Note related issues [4080](https://github.com/trezor/trezor-firmware/issues/4080) and [3663](https://github.com/trezor/trezor-firmware/issues/3663)
  
**Firmware version and revision**
2.8.3

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Safe 5
 - OS: Ubuntu 22.04
 - Browser: n/a
 - Version: n/a

**To Reproduce**
Steps to reproduce the behavior:

### 1. We begin by downloading the latest binary from trezor.

    $ wget https://data.trezor.io/firmware/t3t1/trezor-t3t1-2.8.3{,-bitcoinonly}.bin

### 2. We get the hashes for the binaries.

    $ sha256sum *.bin
    9f68696478e09d7bf8b8f5181413d8a5386b37571dc2f5ed8511a24f4c1d35b7  trezor-t3t1-2.8.3.bin
    49996a1a602f08809427cf3b959f3c70eeef1fcfd45f267bd6d058496a4cc37e  trezor-t3t1-2.8.3-bitcoinonly.bin  
    
### 3. We then proceed with building

    $ git clone https://github.com/trezor/trezor-firmware.git
    $ cd trezor-firmware
    $ git checkout core/v2.8.3
    $ bash -c "./build-docker.sh --models T3T1 core/v2.8.3"

The results:

    Built from commit 39565d3970deb9d696217cda793dc467f2a8e3e5

    Fingerprints:
    dfabe8b10368f268cedaa505e284192329f489519dd71feb5fba7d610ef748fc build/core-T3T1/bootloader/bootloader.bin
    0de51126c17cc0ac623800638dc851c0abd5b787cad5f3aa5843ea2c4cf8248a build/core-T3T1/firmware/firmware.bin
    dfabe8b10368f268cedaa505e284192329f489519dd71feb5fba7d610ef748fc build/core-T3T1-bitcoinonly/bootloader/bootloader.bin
    9eaf99a9420d2a3b9377102eb06b938f5a1886ecb06cccde7fd3cb7a39e1abd7 build/core-T3T1-bitcoinonly/firmware/firmware.bin

### 4. We then begin the process for signature zeroing. We have tried several variations, based on this [closed GitHub issue.](https://github.com/trezor/trezor-firmware/issues/4080)

    $ cp ../trezor-t3t1-2.8.3-bitcoinonly.bin trezor-t3t1-2.8.3-bitcoinonly.bin.zeroed
    $ cp ../trezor-t3t1-2.8.3.bin trezor-t3t1-2.8.3.bin.zeroed
    $ vendorHeaderSize=1024
    $ seekSize=$(( 5567 - vendorHeaderSize ))
    $ count=128
    $ dd if=/dev/zero of=trezor-t3t1-2.8.3.bin.zeroed bs=1 seek=$seekSize count=$count conv=notrunc
    $ dd if=/dev/zero of=trezor-t3t1-2.8.3-bitcoinonly.bin.zeroed bs=1 seek=$seekSize count=$count conv=notrunc

Then: 

    128+0 records in
    128+0 records out
    128 bytes copied, 0.0003931 s, 326 kB/s
    128+0 records in
    128+0 records out
    128 bytes copied, 0.000358158 s, 357 kB/s

    $ sha256sum *.zeroed build/core-T3T1{,-bitcoinonly}/firmware/firmware.bin | sort

We surmise that the header size for the Trezor Safe 5 may be different which have lead to different values:

[... 127 lines cut ...]
+000011d0: 2948 46f0 37f9 7068 fff7 9cfe 022f 0446  )HF.7.ph...../.F
+000011e0: 05d9 b368 1e2b 04bf 65f0 7f05 edb2 44ea  ...h.+..e.....D.
+000011f0: 0520 0025 84b2 2a46 2946 2046 0df1 0203  . .%..*F)F F....
+00001200: adf8 0250 02f0 74f8 b0f1 aa3f 29d1 bdf8  ...P..t....?)...
+00001210: 0260 46b3 1022 2946 01a8 99f0 07fa 3146  .`F..")F......1F
+00001220: 01a8 48f0 11f8 2046 bdf8 0820 0399 0df1  ..H... F... ....
+00001230: 0203 02f0 5df8 b0f1 aa3f 05d0 01a8 48f0  ....]....?....H.
 00001240: 55f8 0e49 0e48 c4e7 0e48 01a9 41f0 5afb  U..I.H...H..A.Z.
 00001250: 074b 1a68 059b 5a40 4ff0 0003 05d0 4ff0  .K.h..Z@O.....O.
 00001260: 83fe 0620 f4e7 0848 f2e7 07b0 f0bd 00bf  ... ...H........
 00001270: ac87 0030 10cd 0e0c 908f 110c 74cd 0e0c  ...0........t...
 00001280: bc90 110c 14a6 110c 04a6 110c 30b5 254d  ............0.%M
 00001290: 00eb c000 244b 05eb 4005 87b0 0c46 2846  ....$K..@....F(F
 000012a0: 0021 1b68 0593 4ff0 0003 03f0 53f8 b0f1  .!.h..O.....S...
```

**Expected behavior**
Hashes should match.

**Screenshots**
n/a

**Additional context**
Requesting documentation changes to this page to include Trezor Safe 5.
https://github.com/trezor/trezor-firmware/blob/main/docs/common/reproducible-build.md

matejcik, 2024-10-11:
@xrviv please try with the updated instructions in #4255

xrviv, 2024-10-14:
Testing. Looks promising.

PR #4255 (2024-10-29): docs: update reproducibility instructions
fixes #3418
fixes #4254 

@bosomt please try reproducible builds based on these instructions
@Hannsek please review in terms of readability
closed 2024-10-29 by matejcik