trezor/trezor-firmware #4331: German translation confirm passphrase overlays confirm button

bosomt, 2024-11-08:
**Describe the bug**
German translation confirm passphrase overlays confirm button

**Steps to reproduce:**
1. Load german translation
2. Enter passphrase on device
3. Observe text and confirm button 

**Info:**
 - Suite version: desktop 24.11.1 (9e9dadff47c5243bf4a10f539a898fe48da46506)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/24.11.1 Chrome/128.0.6613.162 Electron/32.1.2 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.8.5 regular (revision a61e783448043760a7c925e017bd08a4673a61c8)
 - Transport: BridgeTransport 2.0.33

**Screenshots:**
[image]

bieleluk, 2024-11-13:
A very similar overlay issue occurs in the last PIN attempt screen for T2T1:
[image]

bieleluk, 2024-11-14:
Another two occurrences in Portuguese, in the 1st PIN attempt (16 remaining) for T2T1 and T3T1
[image]
[image]

obrusvit, 2026-02-21:
Still valid, it's `passphrase__title_enter`

Thalarion, 2026-09-28:
Enter passphrase on Delizia is problematic in PTBR as well

[image]

closed 2026-10-07 by obrusvit