trezor/trezor-firmware #4371: [Improvment] - Add support for Solana message signing

ffrediani, 2024-11-21:
I have been in search for a wallet that works with Trezor to manage asset in Solana blockchain. I tested most of the main ones that are known to have support for Solana (OKX, Phantom, Solfrale, Nu.FI and Backpack).
Some don't have Trezor support and some have, but partially. When tried to sign a message a got a error "WalletSignMessageError: Not implemented"

When reached to the wallet's support they provided a comprehensive explanation:

"_This is something that Trezor needs to develop. In a nutshell, Trezor doesn't support message signing for Dapp transactions (neither does Ledger, actually, but Dapps find a workaround that they haven't yet implemented for Trezor)._"

Is it something in the radar that can be reached anytime soon ?

As a result in order to deal with Dapps it is necessary to move asset from Trezor to a Hot Wallet that can interact with Dapps on Solana which reduces the security level

ffrediani, 2024-11-25:
Trezor Support confirmed Trezor doesn't support this feature for Solana and asked to use a Hot Wallet without Trezor.

0xbuts, 2024-12-18:
I have the same problem, and received a similar response from support.

I believe this is much more than an improvement, because is blocking users from using DeFi and lots of dApps on Solana securely with their Trezor. 

It would be great if a developer could add some labels to increase the perceived priority of this issue.

monkeym4ster, 2025-05-22:
I need this feature as well. Please let me know of any updates.

ffrediani, 2025-05-27:
@0xbuts agree. It became more than a improvement given these restrictions when interacting with certain dApps on Solana.

matejcik, 2025-07-14:
hello all,

Solana **does not have a message signing specification**.

In the absence of a standard, developers have started to implement a so-called "message signing" by just straight up applying Solana's signature algorithm to random data. This is [kind of endorsed](https://solana.com/developers/cookbook/wallets/sign-message) by the Solana core devs.

Unfortunately, doing things this way is **extremely insecure**, because it allows **every signed message** to be understood as a **transaction** on the Solana network.
Specifically:
* you could pass a transaction to the "Sign message" call and the user would not be aware that they are signing a transaction
* a clever enough hacker may be able to devise a _readable text_ that is at the same time a valid transaction stealing your tokens or something.

Given the above, we **can not securely implement** Solana message signing on a hardware wallet until a proper message signing standard is specified.
(if a competing hardware wallet claims support for Solana message signing, I advise _extreme caution_ with what you are signing using that feature.)

closed 2025-07-14 by matejcik

ffrediani, 2025-07-14:
What a pity, so long for this close.
@matejcik it is understandable what you said and thanks for the details, but it doesn't matter much that much if this means Trezor customers will continue to be unable to "sign messages" on certain dApps and limit their Trezor usage. That's all most will see and talk about they "Cannot use my Trezor fully".
After you explanation, I wonder what makes some dApps implement this requirement ?

If a complete secure Solana message signing can't be implemented right now, some workaround may be considered, with relevant warnings, such as your main competitors made so far, and work alongside with those who establish Solana standards to have a permanent standard later on.

Hannsek, 2025-07-15:
We do not want to allow blind signing at all. 

Once a sane standard will be established, we'll definitely reevaluate.

matejcik, 2025-07-15:
For completeness: we've been made aware of [this specification](https://github.com/solana-labs/solana/blob/master/docs/src/proposals/off-chain-message-signing.md) that would fulfil the requirements for Trezor implementation.
So that option is on the table, depending on how popular this spec is.

Either way, it is not on the roadmap right now.

ffrediani, 2026-03-16:
It has been a difficult thing to convince Trezor team to do some things, some pretty obvious and no brainer sometimes.
Sometimes there is a feeling that if someone brings up something that hasn't been implemented yet, someone will respond that if it was like that someone from the team would have thought or done already.

Hopefully there is a attitude change and this necessary feature gets implemented somehow looking towards user demand and simplicity.