trezor/trezor-firmware #4401: Debuglink response can get cut off if a workflow ends

matejcik, 2024-11-28:
**Describe the bug**
This does not usually happen on emulator because sending data on emulator is very fast.

However, if sending data is _not_ fast, the following race condition may occur:

1. host asks for `DebugLinkGetState`, where the response needs multiple USB packets (this is most screens actually, the layout json is long)
2. firmware starts streaming the `DebugLinkState` response, pausing on every write until the USB link is ready
3. during the pause, wirelink workflow handler is resumed
4. wirelink workflow handler exits and starts sending a Success message, which fits in just one packet
5. Success packet write loop ends (the packet is in an outgoing buffer or w/e)
6. wirelink calls `loop.clear()` to clean out the memory and "reboot" the micropython env
7. this clears out the pending debuglink write task
8. host gets stuck waiting for remaining packets of the `DebugLinkState` response which will never be sent

A hot-fix on host side is to implement a timeout for the DebugLinkState wait, and ignore failures here.

Proper fix on firmware side involves something like registering "units of work" with the workflow manager (and/or loop scheduler), and run `loop.clear()` not after a workflow but instead "when no units of work are pending".
(This approach would also solve the problems that THP is facing with not knowing very well when it's safe to call `loop.clear()`)

mmilata, 2024-12-02:
On current main + #4375 I seem to be able to reproduce it often with `PYTEST_TIMEOUT=24 pytest -v ../tests/device_tests/ -k "test_load_device_utf"`.

romanz, 2025-02-05:
Using #4568 seems to resolve the flakiness of `test_load_device_utf` test:
```
 pytest tests/device_tests/test_msg_loaddevice.py::test_load_device_utf -v --count 10
=================================================================== test session starts ====================================================================
platform linux -- Python 3.10.10, pytest-8.3.2, pluggy-1.5.0 -- /home/rzeyde/src/trezor-firmware/.venv/bin/python
cachedir: .pytest_cache
hypothesis profile 'default' -> database=DirectoryBasedExampleDatabase('/home/rzeyde/src/trezor-firmware/.hypothesis/examples')
Test order randomisation NOT enabled. Enable with --random-order or --random-order-bucket=<bucket_type>
rootdir: /home/rzeyde/src/trezor-firmware
configfile: setup.cfg
plugins: xdist-2.5.0, flaky-3.8.1, forked-1.4.0, hypothesis-6.36.1, ordering-0.6, repeat-0.9.3, timeout-2.1.0, random-order-1.0.4
collected 10 items                                                                                                                                         

tests/device_tests/test_msg_loaddevice.py::test_load_device_utf[1-10] PASSED                                                                         [ 10%]
tests/device_tests/test_msg_loaddevice.py::test_load_device_utf[2-10] PASSED                                                                         [ 20%]
tests/device_tests/test_msg_loaddevice.py::test_load_device_utf[3-10] PASSED                                                                         [ 30%]
tests/device_tests/test_msg_loaddevice.py::test_load_device_utf[4-10] PASSED                                                                         [ 40%]
tests/device_tests/test_msg_loaddevice.py::test_load_device_utf[5-10] PASSED                                                                         [ 50%]
tests/device_tests/test_msg_loaddevice.py::test_load_device_utf[6-10] PASSED                                                                         [ 60%]
tests/device_tests/test_msg_loaddevice.py::test_load_device_utf[7-10] PASSED                                                                         [ 70%]
tests/device_tests/test_msg_loaddevice.py::test_load_device_utf[8-10] PASSED                                                                         [ 80%]
tests/device_tests/test_msg_loaddevice.py::test_load_device_utf[9-10] PASSED                                                                         [ 90%]
tests/device_tests/test_msg_loaddevice.py::test_load_device_utf[10-10] PASSED                                                                        [100%]

============================================================== 10 passed in 201.11s (0:03:21) ==============================================================
```
(tested with TS5 device, on 9d1d062)

romanz, 2025-02-05:
Also, tested with `tests/device_tests/binance/test_get_address.py` (which has reproduced this issue too):
```
$ pytest tests/device_tests/binance/test_get_address.py -v --count 10
==================================================================================== test session starts =====================================================================================
platform linux -- Python 3.10.10, pytest-8.3.2, pluggy-1.5.0 -- /home/rzeyde/src/trezor-firmware/.venv/bin/python
cachedir: .pytest_cache
hypothesis profile 'default' -> database=DirectoryBasedExampleDatabase('/home/rzeyde/src/trezor-firmware/.hypothesis/examples')
Test order randomisation NOT enabled. Enable with --random-order or --random-order-bucket=<bucket_type>
rootdir: /home/rzeyde/src/trezor-firmware
configfile: setup.cfg
plugins: xdist-2.5.0, flaky-3.8.1, forked-1.4.0, hypothesis-6.36.1, ordering-0.6, repeat-0.9.3, timeout-2.1.0, random-order-1.0.4
collected 40 items                                                                                                                                                                           

tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/0-bnb1hgm0p7khfk85zpz5v0j8wnej3a90w709vhkdfu-1-10] PASSED                                     [  2%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/0-bnb1hgm0p7khfk85zpz5v0j8wnej3a90w709vhkdfu-2-10] PASSED                                     [  5%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/0-bnb1hgm0p7khfk85zpz5v0j8wnej3a90w709vhkdfu-3-10] PASSED                                     [  7%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/0-bnb1hgm0p7khfk85zpz5v0j8wnej3a90w709vhkdfu-4-10] PASSED                                     [ 10%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/0-bnb1hgm0p7khfk85zpz5v0j8wnej3a90w709vhkdfu-5-10] PASSED                                     [ 12%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/0-bnb1hgm0p7khfk85zpz5v0j8wnej3a90w709vhkdfu-6-10] PASSED                                     [ 15%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/0-bnb1hgm0p7khfk85zpz5v0j8wnej3a90w709vhkdfu-7-10] PASSED                                     [ 17%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/0-bnb1hgm0p7khfk85zpz5v0j8wnej3a90w709vhkdfu-8-10] PASSED                                     [ 20%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/0-bnb1hgm0p7khfk85zpz5v0j8wnej3a90w709vhkdfu-9-10] PASSED                                     [ 22%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/0-bnb1hgm0p7khfk85zpz5v0j8wnej3a90w709vhkdfu-10-10] PASSED                                    [ 25%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/1-bnb1egswqkszzfc2uq78zjslc6u2uky4pw46x4rstd-1-10] PASSED                                     [ 27%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/1-bnb1egswqkszzfc2uq78zjslc6u2uky4pw46x4rstd-2-10] PASSED                                     [ 30%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/1-bnb1egswqkszzfc2uq78zjslc6u2uky4pw46x4rstd-3-10] PASSED                                     [ 32%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/1-bnb1egswqkszzfc2uq78zjslc6u2uky4pw46x4rstd-4-10] PASSED                                     [ 35%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/1-bnb1egswqkszzfc2uq78zjslc6u2uky4pw46x4rstd-5-10] PASSED                                     [ 37%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/1-bnb1egswqkszzfc2uq78zjslc6u2uky4pw46x4rstd-6-10] PASSED                                     [ 40%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/1-bnb1egswqkszzfc2uq78zjslc6u2uky4pw46x4rstd-7-10] PASSED                                     [ 42%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/1-bnb1egswqkszzfc2uq78zjslc6u2uky4pw46x4rstd-8-10] PASSED                                     [ 45%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/1-bnb1egswqkszzfc2uq78zjslc6u2uky4pw46x4rstd-9-10] PASSED                                     [ 47%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address[m/44h/714h/0h/0/1-bnb1egswqkszzfc2uq78zjslc6u2uky4pw46x4rstd-10-10] PASSED                                    [ 50%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address_chunkify_details[m/44h/714h/0h/0/0-bnb1hgm0p7khfk85zpz5v0j8wnej3a90w709vhkdfu-1-10] PASSED                    [ 52%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address_chunkify_details[m/44h/714h/0h/0/0-bnb1hgm0p7khfk85zpz5v0j8wnej3a90w709vhkdfu-2-10] PASSED                    [ 55%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address_chunkify_details[m/44h/714h/0h/0/0-bnb1hgm0p7khfk85zpz5v0j8wnej3a90w709vhkdfu-3-10] PASSED                    [ 57%]
[... cut ...]
tests/device_tests/binance/test_get_address.py::test_binance_get_address_chunkify_details[m/44h/714h/0h/0/1-bnb1egswqkszzfc2uq78zjslc6u2uky4pw46x4rstd-9-10] PASSED                    [ 97%]
tests/device_tests/binance/test_get_address.py::test_binance_get_address_chunkify_details[m/44h/714h/0h/0/1-bnb1egswqkszzfc2uq78zjslc6u2uky4pw46x4rstd-10-10] PASSED                   [100%]

=============================================================================== 40 passed in 347.28s (0:05:47) ===============================================================================
```

closed 2025-02-07 by romanz