trezor/trezor-firmware #4465: Display flashes old content before refreshing

matejcik, 2025-01-02:
**QA NOK**

im sorry but i still see minor artefact after confirming xpub or receive address
see new video i recorded https://photos.app.goo.gl/FgAREUos9NJGbPo86

**Info:**
 - Suite version: desktop 25.1.0 (20a35a51ea92edb7ffa42c696f6da297c66d2971)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuiteDev/25.1.0 Chrome/128.0.6613.186 Electron/32.2.6 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.8.7 regular (revision a8e5d471fe35164cbf2529a2c0042ca465f6f159)
 - Transport: BridgeTransport 2.0.33

_Originally posted by @bosomt in https://github.com/trezor/trezor-firmware/issues/4443#issuecomment-2567725889_

Hannsek, 2025-01-04:
Can we fix it and cherry pick it into January release?

closed 2025-01-23 by TychoVrahe

bosomt, 2025-02-03:
**QA OK**

 - Device: Trezor T3T1 2.8.8 regular (revision bbcb920213b0ab847a2fedf825eac70652e33279)

PR #4492 (2025-01-23): fix fading issues
fixes #4392 fixes #4465 

The slow fade in bootloader was due to function raising backlight was taking too long, waiting for display TE signal each time. Waiting for TE is removed as its no longer needed due to different mechanism using pending update - so now we will normally wait only once, after the display content is changed. In addition, the backlight fading function is rewritten to ensure constant fading time, at the possible expense of smoothness.

To mitigate the residual image issue, additional wait cycle is added when raising backlight.

As part of this fix i did i  [... cut]