trezor/trezor-firmware #4474: not able to install poetry with python 3.13

bosomt, 2025-01-07:
```
❯ poetry install
The "poetry.dev-dependencies" section is deprecated and will be removed in a future version. Use "poetry.group.dev.dependencies" instead.
Installing dependencies from lock file

Package operations: 56 installs, 0 updates, 0 removals

  - Installing cffi (1.15.0): Failed

PEP517 build of a dependency failed
```

whole error https://pastebin.com/b9eVASPm

prusnak, 2025-01-07:
reopened to have the PR #4476 tested by @bosomt

prusnak, 2025-01-17:
there is now another fix in https://github.com/trezor/trezor-firmware/pull/4513

bosomt, 2025-01-17:
pillow fixed, 
not sure what to do with trezor-firmware (0.0.0) ;)
i will retest after merge

```
❯ poetry install
The "poetry.dev-dependencies" section is deprecated and will be removed in a future version. Use "poetry.group.dev.dependencies" instead.
Installing dependencies from lock file

Package operations: 35 installs, 0 updates, 0 removals

  - Installing shamir-mnemonic (0.3.0)
  - Installing slip10 (1.0.1)
  - Installing pillow (11.1.0)
  - Installing autoflake (2.3.1): Installing...
  - Installing autoflake (2.3.1)
  - Installing black (24.3.0)
  - Installing binsize (0.1.3)
  - Installing demjson3 (3.0.5)
  - Installing coverage (4.5.4)
  - Installing dominate (2.6.0)
  - Installing fido2 (0.8.1)
  - Installing graphviz (0.19.1)
  - Installing flake8-annotations (3.1.1)
  - Installing flaky (3.8.1)
  - Installing flake8-requirements (2.1.0)
  - Installing hypothesis (6.36.1)
  - Installing inotify (0.2.10)
  - Installing mako (1.1.6)
  - Installing munch (2.5.0)
  - Installing nanopb (0.4.5.post1)
  - Installing pyasn1 (0.4.8)
  - Installing pyserial (3.5)
  - Installing pytest-ordering (0.6)
  - Installing pytest-random-order (1.0.4)
  - Installing pytest-timeout (2.1.0)
  - Installing pytest-xdist (2.5.0)
  - Installing python-bitcoinlib (0.11.0)
  - Installing scan-build (2.0.20)
  - Installing scons (4.8.0)
  - Installing towncrier (23.11.0)
  - Installing tox (3.24.5)
  - Installing vulture (2.6)
  - Installing yamllint (1.26.3)
  - Installing toiftool (0.1.0 /Users/bosomt/Trezor/trezor-firmware/python/tools/toiftool)
  - Installing trezor-core-tools (0.1.1 /Users/bosomt/Trezor/trezor-firmware/core/tools)
  - Installing trezor-pylint-plugin (0.1.0 /Users/bosomt/Trezor/trezor-firmware/tools/trezor-pylint-plugin)

Installing the current project: trezor-firmware (0.0.0)
Error: The current project could not be installed: No file/folder found for package trezor-firmware
If you do not want to install the current project use --no-root.
If you want to use Poetry only for dependency management but not for packaging, you can disable package mode by setting package-mode = false in your pyproject.toml file.
If you did intend to install the current project, you may need to set `packages` in your pyproject.toml file.
```

bosomt, 2025-01-29:
Still failing even after https://github.com/trezor/trezor-firmware/pull/4513 merge
not sure if related or not /i have fresh macos installation/

```
Installing the current project: trezor-firmware (0.0.0)
Error: The current project could not be installed: No file/folder found for package trezor-firmware
If you do not want to install the current project use --no-root.
If you want to use Poetry only for dependency management but not for packaging, you can disable package mode by setting package-mode = false in your pyproject.toml file.
If you did intend to install the current project, you may need to set `packages` in your pyproject.toml file.

bosomt@Tomass-MacBook-Air trezor-firmware %
```

matejcik, 2025-01-29:
i believe you can ignore this error, your env should be already set up.
if not, specify `--no-root`.

we can't set the configuration option as recommended just yet because we're on a range of poetry versions some of which don't support it, so that's a separate issue, but AFAIK this is not blocking your usage