trezor/trezor-firmware #4498: Create Multi-share Backup auto abort after dry run

bosomt, 2025-01-14:
**Describe the bug**
Its not possible to Create Multi-share Backup.
After dry run modal in Suite closes, reproduced on multiple platforms and TS5 devices.
I will try to find if its related to Suite version or firmware version. 

**Steps to reproduce:**
1. Navigate to Settings/device and press Create Multi-share Backup
2. Go thru dry run 
3. Observe SUite behavior

**Info:**
 - Suite version: desktop 25.1.1 (11cab04dfc05ddb63cb6279bbc289994bf6c0583)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/25.1.1 Chrome/128.0.6613.186 Electron/32.2.6 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.8.7 regular (revision 8a254aa8eae82f99630df63f40e4d290066a3efc)
 - Transport: BridgeTransport 3.0.0-bundled.25.1.1

**Screenshots:**

https://github.com/user-attachments/assets/a3769a57-f681-4faa-bcfe-2db58c576251

Hannsek, 2025-01-14:
Is onboarding to multi-share backup affected as well?

bosomt, 2025-01-14:
**QA OK**

i was not able to reproduce it on production version of Suite 🤔 

**Info:**
 - Suite version: web 24.12.3 (b6e22484a37d0a2197338615085aea25742fa0f3)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.8.7 regular (revision 8a254aa8eae82f99630df63f40e4d290066a3efc)
 - Transport: WebUsbTransport

bosomt, 2025-01-14:
**QA OK**

i was not able to reproduce it on webUSB connection, looks like its caused by node bridge / i will check bridge 2.0 ASAP /

**Info:**
 - Suite version: web 25.1.1 (11cab04dfc05ddb63cb6279bbc289994bf6c0583)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.8.7 regular (revision 8a254aa8eae82f99630df63f40e4d290066a3efc)
 - Transport: WebUsbTransport

bosomt, 2025-01-14:
**QA OK**

with bridge 2.0 i was not able to reproduce issue

**Info:**
 - Suite version: desktop 25.1.1 (11cab04dfc05ddb63cb6279bbc289994bf6c0583)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/25.1.1 Chrome/128.0.6613.186 Electron/32.2.6 Safari/537.36
 - OS: MacIntel
 - Screen: 1512x982
 - Device: Trezor T3T1 2.8.7 regular (revision 8a254aa8eae82f99630df63f40e4d290066a3efc)
 - Transport: BridgeTransport 2.0.33