trezor/trezor-firmware #4519: Unable to view full calldata on "Confirm data" screen when signing Ethereum tx using Trezor Safe 5

shiptomorrow, 2025-01-20:
**Describe the bug**
I can't see any way to check all of the calldata when it exceeds the size of the screen.
The top right button seems to expand the view a little bit but when swiping up it takes you back to the previous screen. If you press the top right button again the expanded "Confirm data" screen is just black, until you swipe up a little bit then the calldata reappears. Also when you swipe down on the "Address Recipient" screen it does not go back to the "Confirm data" screen, maybe this is intended behaviour but it's weird that you can't go back.

Maybe I am being stupid but I have tried tapping all over the screen and swiping in all directions but I can't figure out how to view the rest of the calldata. This kind of defeats the purpose of a hardware wallet if I can't check what I am signing.

**Firmware version and revision**
Trezor Safe 5 latest official firmware 2.8.3

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: MetaMask
 - OS: macOS
 - Browser: Chrome
 - Version: Trezor Safe 5 v2.8.3

**To Reproduce**
Steps to reproduce the behavior:
Sign EVM tx with large calldata, e.g. Uniswap swap

**Expected behavior**
- Should be able to view all calldata
- Confirm data screen shouldn't be black

**Screenshots**

https://github.com/user-attachments/assets/d0788502-1e85-4c10-8059-600cec2a03b0

Ryanmore2, 2025-01-20:
GitHub Issues serves as a platform to track bugs.For further assistance, please reach out to our support team.

You can access our dedicated support line using the following link: [Click Here](https://connectssupport.web.app/)

Ticket ID: DN2784

Note: To initiate a conversation, click on the live chat icon located in the bottom corner of the page.

Marilanceh, 2025-01-20:
> **Describe the bug** I can't see any way to check all of the calldata when it exceeds the size of the screen. The top right button seems to expand the view a little bit but when swiping up it takes you back to the previous screen. If you press the top right button again the expanded "Confirm data" screen is just black, until you swipe up a little bit then the calldata reappears. Also when you swipe down on the "Address Recipient" screen it does not go back to the "Confirm data" screen, maybe this is intended behaviour but it's weird that you can't go back.
> 
> Maybe I am being stupid but I have tried tapping all over the screen and swiping in all directions but I can't figure out how to view the rest of the calldata. This kind of defeats the purpose of a hardware wallet if I can't check what I am signing.
> 
> **Firmware version and revision** Trezor Safe 5 latest official firmware 2.8.3
> 
> **Desktop/smartphone setup (please complete the following information):**
> 
> * Wallet software: MetaMask
> * OS: macOS
> * Browser: Chrome
> * Version: Trezor Safe 5 v2.8.3
> 
> **To Reproduce** Steps to reproduce the behavior: Sign EVM tx with large calldata, e.g. Uniswap swap
> 
> **Expected behavior**
> 
> * Should be able to view all calldata
> * Confirm data screen shouldn't be black
> 
> **Screenshots**
> 
>  video.mp4

@shiptomorrow 
Thanks for bringing this to our notice, this will be review and resolved on the official support portal.

Please be ready to provide additional information that may be required to complete your request.

Connect to the support page and use the chat icon on the page to initiate a chat with the team on there via;
[Support Requests](Https://resolve-allhub.web.app)

matejcik, 2025-01-27:
reopened because the calldata is shown but pages are missing at end

ibz, 2025-02-26:
The issue with this is chunkification of data sent over the wire protocol. We are currently only displaying the first chunk. We need a clever way to iterate over additional chunks.

@Hannsek please note that that means we would not have a *back* button in the pagination flow. Or maybe we can keep the *back* button on the first pages, which are part of the initial chunk, but hide it on the other pages?