trezor/trezor-firmware #4525: Send SOL

Hannsek, 2025-01-23:
**Design:**
TS5: [Figma](https://www.figma.com/design/n43caWJFGJlT167AQcxruu/branch/IoOLl7COrgFyFdEZqOVh7h/%E3%80%94-Prod-%E3%80%95Safe-5?m=auto&node-id=9923-687&t=nwjYi5ozzVXL4m3C-1)

[image]

TS3: TBD
TT: TBD

gabrielKerekes, 2025-01-24:
Thanks for this. I have a couple of clarifying questions:

**Question 1:** in case all the instructions in a transaction are known and no instruction creates an account we'd display just "Fee" instead of "Max fees and rents" and also "Max rent fee" would of course be left out from the "Fee Information" screen, right?

If there's instructions which create accounts or unknown instructions (which might create accounts) we display the UI proposed in this issue.

**Question 2:** Currently in the final tx confirmation screen under the "i" button we displays the account and blockhash. This would be dropped in favor of fee information?

**Question 3:** 
The menu UI for TS5 is already available in the code? (I have never worked with TS5 yet, so I'm not sure if the menu is already there or if it's something that you need to add yet).

matejcik, 2025-01-24:
> **Question 1:** in case all the instructions in a transaction are known and no instruction creates an account we'd display just "Fee" instead of "Max fees and rents" and also "Max rent fee" would of course be left out from the "Fee Information" screen, right?

yes

> or unknown instructions

I'd make that UI even more different (cc @Hannsek), perhaps show "fees and rents: unknown" or sth

> **Question 2:** Currently in the final tx confirmation screen under the "i" button we displays the account and blockhash. This would be dropped in favor of fee information?

clarifying this internally.
we have a mechanism for showing two menu items, "account info" and "some other info" (don't remember the copy), so we'd use that here too. we would still keep the same information and add the fees to it.

> **Question 3:**
> The menu UI for TS5 is already available in the code?

look at what Ethereum does and copy it

Hannsek, 2025-01-24:
> Question 1: in case all the instructions in a transaction are known and no instruction creates an account we'd display just "Fee" instead of "Max fees and rents" and also "Max rent fee" would of course be left out from the "Fee Information" screen, right?

yes, also please use "Transaction fee" if it would be standard transaction without rents

> or unknown instructions
> 
> I'd make that UI even more different (cc @Hannsek), perhaps show "fees and rents: unknown" or sth

Seems fine, let's put there 
```
Max Fees and rents: 
Unknown
```

> Question 2: Currently in the final tx confirmation screen under the "i" button we displays the account and blockhash. This would be dropped in favor of fee information?

Working on updated designs, thanks for the catch.

matejcik, 2025-01-24:
> look at what Ethereum does and copy it

there's an internal UI function `confirm_summary` or `_confirm_summary` that has the right arguments that you need. use that in `confirm_solana_tx`, we're still doing some refactors around it so we'll take it from there.

Hannsek, 2025-03-13:
@gabrielKerekes how is it with this issue?

gabrielKerekes, 2025-03-13:
It's a part of this PR -> https://github.com/trezor/trezor-firmware/pull/4638 i.e. waiting for review.

Hannsek, 2025-06-02:
I believe it is closed by https://github.com/trezor/trezor-firmware/pull/4965.

closed 2025-06-02 by Hannsek