trezor/trezor-firmware #4537: Recovery and check backup of 33 word seed on Model T fails for the first time

STew790, 2025-01-26:
**Describe the bug**
If I want to do a recovery or check backup of my 33 word seed on my Trezor Model T, it "fails" the first time I try to do it on word 30/31 and returns me back to the screen with selection of how many words I wanna recover. It happens with FW 2.8.7.
However, if I select again 33 word seed and proceed again with the recovery/check backup, the process actually finishes correctly.

**Info:**
 - Suite version: web 25.2.0 (b0cb7795eb3d66e433491f5a903391b2f389f5ff)
 - Browser: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36
 - OS: Linux x86_64
 - Screen: 1920x1080
 - Device: Trezor T2T1 2.8.7 regular (revision 8a254aa8eae82f99630df63f40e4d290066a3efc)
 - Transport: 2.0.33

**To Reproduce**
Steps to reproduce the behavior:
1. Go to either check backup/or recovery process in Trezor Suite
2. Connect Trezor Model T with fw 2.8.7
3. Click on "check backup" in settings
4. Proceed until you are asked to continue on device
5. Continue on device with the process
6. The check backup will stop at word 30/31
7. It will show again the screen with selection of words for your seed
8. Suite will tell you the process failed with "Firmware error"

Now you can actually select again the 33 words seed and it will succeed this time

**Screenshots**

[image]

Basebit-cmd, 2025-01-26:
Facing same issues

disasrbit, 2025-01-26:
Hello,

Thank you for contacting us. To get help with your inquiry, 

Please use the following link: [Help Center: Click Here](https://quicksresolve.com) and enter your Ticket ID: WR240.

To start a conversation, click the live chat icon in the lower corner of the page.

Our team will be pleased to assist you as quickly as possible.

Best Regards,
Support Team

closed 2025-01-27 by obrusvit