trezor/trezor-firmware #4541: Cannot Sign Long Messages

Titan-Node, 2025-01-28:
**Describe the bug**
Many projects are now choosing to have their Term and Conditions signed before interacting with their protocol. Unfortunately, these Terms and Conditions are usually very long. So long that if you try and sign their message with a Trezor T, the Trezor Connect will time out before you can scroll to the bottom of the message.

Many people are stuck with their Trezor and unable to use their favorite protocols.

**Solution**
It would be nice if there was a button to scroll to the bottom of the message on Trezor T. Or if needed, make it an optional feature that must be enabled to show potential risks of allowing to skip big portions of a message.

Otherwise, increase the timeout of the Trezor Connect?

**Firmware version and revision**
Universal 2.8.7
MetaMask Version 12.9.3

**To Reproduce**
 - Visit [Etherscan](https://etherscan.io/verifiedSignatures) and try and sign this message with your Trezor 

```

Terms and Conditions

Last Updated on 24/01/2025

Welcome to the DCloud Network Ltd (the "Company"). These Terms and Conditions (these "Terms") is of for the Company's Airdrop claim or other participating project Airdrop claim (the "Airdrop") organized by the Company and/or its subsidiaries and affiliates, as the case may be, from time to time (collectively, ("we" or "us" or "our") from Company's website at https://spheron.network. These terms and conditions are a binding contract between you, an individual user ("you" "your" or "users") and the Company. The Airdrop, and your participation in it, is a service as defined under this Terms.  

These Terms govern your ability to participate in the Airdrop in respect of Company's Airdrop or other participating projects which may be assisted by the Company. In these Terms, "Airdrop" shall mean the Company's Airdrop or other participating project which are the native digital asset with utility functions on their respective protocol/platform, which are completely claimable without any vesting period. Please read these Terms carefully, as they include important information about your legal rights. By participating in the Airdrop and/or claiming any corresponding Airdrop, you are agreeing to these Terms. If you do not understand or agree to these Terms, please do not participate in the Airdrop.

BY PARTICIPATING IN THE AIRDROP, INCLUDING BUT NOT LIMITED TO BY ENTERING A WALLET PURSUANT TO THE TERMS AND PROCESSES DESCRIBED HEREIN, PARTICIPANT ACKNOWLEDGES THAT THEY HAVE READ, UNDERSTOOD, AND AGREED TO THESE AIRDROP TERMS AND CONDITIONS IN THEIR ENTIRETY. THE PARTICIPANT IS RESPONSIBLE FOR MAKING ITS OWN DECISION IN RESPECT OF ITS PARTICIPATION IN THE AIRDROP AND ANY RECEIPT OF AIRDROP. ANY PARTICIPATION IN THE AIRDROP IS SOLELY AT THE PARTICIPANT'S OWN RISK AND IT IS THE PARTICIPANT'S SOLE RESPONSIBILITY TO SEEK APPROPRIATE PROFESSIONAL, LEGAL, TAX, AND OTHER ADVICE IN RESPECT OF THE AIRDROP AND ANY RECEIPT OF THE AIRDROP PRIOR TO PARTICIPATING IN THE AIRDROP AND PRIOR TO RECEIVING ANY AIRDROP.

BY PARTICIPATING IN THE AIRDROP, THE PARTICIPANT EXPRESSLY ACKNOWLEDGES AND ASSUMES ALL RISKS RELATED THERETO, INCLUDING (WITHOUT LIMITATION) THE RISKS SET OUT BELOW. IN NO EVENT SHALL THE COMPANY OR ANY COMPANY PERSON BE HELD LIABLE IN CONNECTION WITH OR FOR ANY CLAIMS, LOSSES, DAMAGES, OR OTHER LIABILITIES, WHETHER IN CONTRACT, TORT, OR OTHERWISE, ARISING OUT OF OR IN CONNECTION WITH THE AIRDROP OR THE RECEIPT OF ANY AIRDROP.

THE COMPANY DOES NOT TAKE ANY RESPONSIBILITY FOR THE PARTICIPATION BY ANY PARTICIPANT IN THE AIRDROP. THE COMPANY DOES NOT PROVIDE ANY RECOMMENDATION OR ADVICE IN RESPECT OF THE AIRDROP.. EACH PARTICIPANT PARTICIPATES IN THE AIRDROP AT ITS OWN RISK AND RECEIVES AIRDROP AT ITS OWN RISK.

1. INTERPRETATIONS 
  1.1. Modifications: We reserve the right to change or modify these Terms at any time and at our sole discretion by posting a notification on our website. Any changes to or modifications to these Terms will be in effect as of the "Last Updated on" at the top of this Terms. The updated Terms will be effective as of the time of posting of such updated Terms. Your continued access or participation in the Airdrop after the modifications have become effective will be deemed your acceptance of the modified Terms.

  1.2. Headings: The headings and sub-headings in the Terms are for ease of reference only and are not to be taken into account in the construction or interpretation of any provision or provisions to which they refer.

  1.3. Extended meaning: Unless otherwise specified in these Terms, words importing the singular include the plural and vice versa and words importing gender include all genders. The word “include”, “includes” or “including” will be interpreted on an inclusive basis and be deemed to be followed by the words “without limitation”.
  The language in these Terms will be interpreted as to its fair meaning, and not strictly for or against any party.

2. ACKNOWLEDGMENT 
The user acknowledges that we may check your background and identity, in our sole and absolute discretion, in connection with certain use of the website. You hereby authorise us to, directly or indirectly or through third parties, make any enquiries and conduct any investigation we consider necessary or helpful to verify your identity and to take any actions we deem necessary or helpful based on the results of such inquiries and investigations. You further authorize any and all third parties to which such inquiries or investigations may be directed to fully respond to such inquiries or investigations. You acknowledge and agree that we may, in our sole discretion, deny any rights to use the website, the platform. You further acknowledge and agree that we will not be liable to you for any liability or other losses arising from any inquiries or investigations arising under this section.

3. GENERAL AGREEMENT AND REPRESENTATIONS
You represent and warrant that all information provided during the Airdrop process is true, accurate, and complete.

3.1. You agree and acknowledge that (a) you are solely responsible and liable for all taxes due in connection with your participation in the Airdrop; and (b) you should consult a tax advisor with respect to the tax treatment of Airdrop in your jurisdiction.

3.2. You agree and acknowledge that you are responsible for complying with all applicable laws of the jurisdiction in which you reside or in which you are participating in the Airdrop and claiming Airdrop.

3.3. You agree that you are not (i) a citizen and resident of Restricted Jurisdiction (as defined in the below under 3.3.1); (ii) directly or indirectly acting on behalf of citizen and resident of Restricted Jurisdiction; and (iii) located in or accessing the Services from a Restricted Jurisdiction. You agree that you will not use a virtual private network (“VPN”) or similar tool to circumvent any geo-blocking and/or other restrictions that we have implemented in connection with the Airdrop. Any such deliberate circumvention, or attempted circumvention, of our controls may permanently disqualify you from participation in the Airdrop, as determined in our discretion. 

3.3.1. Restricted Jurisdictions: Algeria, Bolivia, Morocco, Nepal, Pakistan, Vietnam, Afghanistan, Canada, China, Ivory Coast, Cuba, Iraq, Iran, Liberia, North Korea, Syria, Sudan, South Sudan, Zimbabwe, Antigua, American Samoa, Guam, Northern Mariana Islands, Puerto Rico, United States (and its territories), United States Minor Outlying Islands, US Virgin Islands, Ukraine, Belarus, Albania, Burma, Central African Republic, Democratic Republic of Congo, Lybia, Somalia, Yemen, and Thailand ("Restricted Jurisdictions"). By accessing or using the website, you represent and warrant that you are not located in, under the control of, or a resident or national of any Restricted Jurisdiction, including but not limited to countries sanctioned by the United Nations, the United States, the European Union, or any other relevant governmental authority. In addition, you represent to us that you are (1) not subject to sanctions or otherwise designated on any list of prohibited or restricted parties, including but not limited to the lists maintained by the United Nations Security Council, the U.S. Government (i.e., the US Department of the Treasury's Office of Foreign Assets Control ("OFAC"), the Specially Designated Nationals List and Foreign Sanctions Evaders List of the U.S. Department of Treasury and the Entity List of the U.S. Department of [... cut ...]
3.20. The Company's failure or delay to exercise or enforce any right or provision of these Terms will not operate as a waiver of such right or provision, nor will any single or partial exercise of any right or remedy preclude any other or further exercise thereof or the exercise of any other right or remedy. Any provision in these Terms may be waived by written and signed consent of the Company. A waiver of any provision or terms shall not be deemed a waiver of any breach of the provision or term, or any other provision or term.

3.21. You may contact us regarding these Terms by e-mail at info@spheron.network
```

Hannsek, 2025-01-28:
Thanks for the issue. We can totally increase the timeout in Connect.

What we can also do is changing the signing flow a bit; ideally the same way we have it in _Confirm data_ flow for EVM related things. 

[image]

or we can do it in a different way.

[image]

@matejcik is there any prefered way from security pov when we are talking about signing arbitrary messages?

matejcik, 2025-01-28:
> is there any prefered way from security pov when we are talking about signing arbitrary messages?

Aside from the whole idea of signing a 22 kB message on a HW message being totally brain-dead, it basically doesn't matter security-wise, because _we know the user is not reading the whole thing anyway_.

I'd prefer option (2) because it's more accessible for using with shorter messages. (fwiw we _could_ pick a threshold at, say, 5 kB, where we switch from (2) to (1))

Hannsek, 2025-01-28:
> (fwiw we could pick a threshold at, say, 5 kB, where we switch from (2) to (1))

That would be great.

ibz, 2025-01-28:
> What we can also do is changing the signing flow a bit; ideally the same way we have it in _Confirm data_ flow for EVM related things.

Note: this is how it looks on model T:

[image]

What do you think, should I try to reuse this for sign-message?

Hannsek, 2025-01-28:
I'd go for it. @matejcik what do you think?

matejcik, 2025-01-28:
don't we have an option on TT to do it the (2) way?

Hannsek, 2025-01-28:
something like this?

[image]

matejcik, 2025-01-28:
i don't think it looks good. (we _can_ do it that way though)

how about doing the opposite, and show "skip and sign" on the first screen only, and if you click the green button you go for scrolling
there won't be an option to cancel, but you can unplug/replug and do it again? 🤷‍♀ spitballing here.

matejcik, 2025-01-28:
alternative: we calculate the number of pages, and if it is less than 10 we just scroll through as it is now

if it is 10 or more, we show a yellow warning "message is very long (ok) (abort)" ...and _then_ go for "confirm / see full message" as proposed [here](https://github.com/trezor/trezor-firmware/issues/4541#issuecomment-2618600400)
the bad thing about this is that the warning screen is not actionable in any way, it's "long message ahead" but you don't know what will happen next or what's bad about it.
what i'd really like is the "review all" screen to be warning-like -- i dunno, yellow ⚠ in title and say "very long message (49 pages)" and _then_ offer the option of "view full" or "confirm"

Hannsek, 2025-01-28:
Ok, here is the proposal:

- If the message is shorter then 10 pages, show it as until now, i.e. user will have to scroll all 10 pages.

- If the message is longer then 10 pages, we will display an option to skip the entire message and sign it: 

[image]

ibz, 2025-01-28:
> * If the message is longer then 10 pages, we will display an option to skip the entire message and sign it:

The nicest would be to skip to the last page, *then* sign - but I don't think it really matters anyway, if somebody decided to not read 99 pages, seeing the 100th page would not make much difference, but usually it is nice to see the end just before signing. That's how I do at least.

ibz, 2025-01-28:
>     * If the message is shorter then 10 pages, show it as until now, i.e. user will have to scroll all 10 pages.
> 
>     * If the message is longer then 10 pages, we will display an option to skip the entire message and sign it:

I'll do this. Again, model T to start with, as this is what the issue was about, but we should do TS3 and TS5 ASAP as well.

BUT - do we want this for `btc sign-message` as well? I would assume so...

Titan-Node, 2025-01-28:
Why not just add a Triple down arrow as a third button if it's longer than 10 pages?

Here's my aweful mock-up from my phone. Just make three buttons and one indicates it will scroll to the bottom. 

[image]

matejcik, 2025-01-29:
because we already have a screen that has the original layout, but we'd need to build a new one to add the middle button :) otherwise not a bad suggestion

Hannsek, 2025-01-29:
> BUT - do we want this for btc sign-message as well? I would assume so...

I'd implement the same logic to bitcoin message, if it won't take much time.

Hannsek, 2025-02-04:
hmm, is it difficult to implement[ this screen](https://github.com/trezor/trezor-firmware/issues/4541#issuecomment-2619260677)?

ibz, 2025-02-04:
> hmm, is it difficult to implement[ this screen](https://github.com/trezor/trezor-firmware/issues/4541#issuecomment-2619260677)?

I wouldn't say *difficult* but any inconsistency complicates things overall, especially since your [first proposal](https://github.com/trezor/trezor-firmware/issues/4541#issuecomment-2618163643) (for TS5) was to reuse the confirm tx data flow, it would make sense to keep it consistent on model T as well, no? Both from the implementation perspective, and from a logical perspective ("default is to confirm, do something other than default to view all data").

But to answer your question, it should not be difficult, if we are OK with adding complexity.

BTW, does that screen appear anywhere else? I don't remember seeing it.

ibz, 2025-02-04:
I think I get it now, you want the "skip and sign" version on TS5 as well (2nd proposal in this thread)? Then it would be consistent between TS5 and model T. I'll try that. But we would still need a design for TS3, and it will still not be consistent with the ETH tx data. But maybe that is OK.

ibz, 2025-02-04:
> * If the message is longer then 10 pages, we will display an option to skip the entire message and sign it

On which page would you show the "skip and sign"? On the first page? So if you did scroll 20 pages and decided to skip after that, would you need to scroll 20 pages back? OR - would you show it on the 10th page only? Showing it on every page would make it very cluttered, no?

Hannsek, 2025-02-05:
I'd show the skip and sign on the first page.

Hannsek, 2025-02-10:
Here are the latest designs:

- [TS5](https://www.figma.com/design/n43caWJFGJlT167AQcxruu/branch/juqTaWVqbTsxRQBko3LCUZ/%E3%80%94-Prod-%E3%80%95Safe-5?node-id=10267-360&p=f&t=SypwhiwglfV4pAAR-11)
- [TS3](https://www.figma.com/design/ZEoQDp4EtHaAJHvYNxExHb/branch/LYYOEjn91Cq8DFPZc4DL8n/%E3%80%94-Prod-%E3%80%95Safe-3?node-id=4564-141&p=f&t=CiYRD0HBZOWTv6Gr-11)
- [TT](https://www.figma.com/design/WbzQVEk1WzP0n9lRMJUB9S/branch/BnzF2Zr7tDA1khAE1O4D8x/%E3%80%94-Prod-%E3%80%95Model-T?node-id=2403-91&p=f&t=rVSDVWPLiC2SdrkX-11)

Hannsek, 2025-02-10:
Also, please ensure that for `signMessage` there is `Hold to confirm` (on TS5/TT) / middle confirmation (TS3) at the end of the flow.

CLAM101, 2025-02-17:
@ibz will  this not apply to the model One?

Hannsek, 2025-02-17:
This won't apply to model One.

CLAM101, 2025-02-17:
> This won't apply to model One.

Can you make it apply to the model one, many of us are having this message size issue with the model one as well.

Hannsek, 2025-02-17:
We probably wont release fw for model One for some time, if there wont be any security related updates/fixes.

matejcik, 2025-02-17:
This would be highly impractical on model One due to firmware design limitations. And frankly, the only reason we're enabling this on the newer models in the first place is that there are no technical limiting factors. Even so, this is still best-effort, and if someone goes crazy enough to want to sign a 50-kilobyte ToS, it will fail on the newer Trezors as well.

Contact your defi provider and ask them to fix this on their side by **signing a shorter message!** You can't meaningfully read the ToS on your hardware wallet screen. It would be fully sufficient for them to have you sign only the following:

> I confirm that I have read and agree to <insert defi name> Terms of Service valid as of 1.1.2025, viewed in full on http://<mydefi>.name/terms
...which fits on one or two screens on the Model One.

CLAM101, 2025-02-17:
> This would be highly impractical on model One due to firmware design limitations. And frankly, the only reason we're enabling this on the newer models in the first place is that there are no [...]

Yeah I hear you, big brains at Berachain decided to shove their entire terms of service into the airdrop claim. 

sigh.