trezor/trezor-firmware #4542: Decode the most used EVM contract methods

Hannsek, 2025-01-28:
Currently, during EVM calls, users see a large number of values that are not easily human-readable. To improve this, we propose parsing the calls and presenting them in a more understandable format.

Until clear signing is implemented, we should focus only on the most commonly used calls and hardcode them into the firmware.

Here is my suggestion for prioritized calls:

`Approve()`
`Revoke()`
I’m unsure about the feasibility of parsing `permit2() `and `execute()` in a meaningful way — can these be parsed clearly?

`transfer()` and `transferFrom()` should already be supported iirc.

matejcik, 2025-01-28:
we only support `transfer()` because `transferFrom()` is not useful for end-users -- it is transferring someone else's token under allowance.

permit2 is interesting: first, it's a regular `Approve()` call with a known destination (the Permit2 contract), which we may want to special-case -- but the user still need to approve each of their tokens for Permit2 separately

then there's an off-chain permit2 message which seems to be an EIP712 structure signable by `EthereumSignTypedData` -- we can do that today but a nicer representation would be helpful
...and then there's some other, possibly contract-specific? call that consumes the signature of this message.
so to make this extra nice, we'd have to integrate both those two things under a single confirmation, and that would be quite a lot of work.

but to only make it _work_ it seems we don't have to do any additional work except for supporting `Approve()`

prusnak, 2025-03-04:
related: https://github.com/trezor/trezor-firmware/issues/69

tomasklim, 2025-03-13:
Revoke is Approve with 0 allowance

closed 2025-05-21 by ibz