trezor/trezor-firmware #4560: Sol Stake: Align FW and Suite when signing

tomasklim, 2025-01-30:
### User story
As a User trying to stake/unstake/claim SOL
I want to confirm the staking/unstaking/claim transaction on my Trezor device

### Requirements
- Identify and decode the following Solana staking transactions in the FW
  - stake, unstake, claim
  - how to identify these transactions: see bellow
- Remove the redundant screens on the Trezor FW, when signing the relevant SOL staking TX
  - currently there are around 8 steps when signing the Stake transaction - we should reduce these steps as much as possible and ideally present a similar flow as with ETH staking (see screenshot bellow) 
- Design
  -  TBD - should we say "Stake with Everstake" on the FW or should we keep the message generic and dont identify Evestake.  

[image]

see ETH implementation: 
https://github.com/trezor/trezor-suite/issues/11583

### Identifying staking trsansactions
Code snippet:
function getTransactionStakeType(tx: SolanaValidParsedTxWithMeta): StakeType | undefined {
    const { instructions } = tx.transaction.message;

    if (!instructions) {
        throw new Error('Invalid transaction data');
    }

    for (const instruction of instructions) {
        if (instruction.programId === STAKE_PROGRAM_PUBLIC_KEY && 'parsed' in instruction) {
            const { type } = instruction.parsed || {};

            if (type === 'delegate') return 'stake';
            if (type === 'deactivate') return 'unstake';
            if (type === 'withdraw') return 'claim';
        }
    }

    return undefined;
}
The provided function, getTransactionStakeType, is designed to extract and identify the stake type of a given Solana transaction. This stake type corresponds to specific actions within the Solana staking program, such as staking, unstaking, or claiming rewards. 

Here's how the decoding process works:

Key Steps in the Method:
1. Accessing Transaction Instructions:
- The function begins by accessing the instructions property from the transaction's message object (tx.transaction.message.instructions).
- If no instructions are present, an error is thrown, as this indicates invalid transaction data.
2. Iterating Through Instructions:
- Each instruction in the transaction is inspected in sequence.
3. Checking for the Stake Program:
- The function identifies instructions related to the Solana staking program by matching the programId against the known STAKE_PROGRAM_PUBLIC_KEY.
4. Extracting and Interpreting Parsed Data:
- For instructions belonging to the staking program, the function ensures they contain a parsed field.
- The parsed.type field is then analyzed:
  - "delegate" maps to the stake action.
  - "deactivate" maps to the unstake action.
  - "withdraw" maps to the claim action.
5. Returning the Stake Type:
- If a matching instruction is found, the corresponding stake type (stake, unstake, or claim) is returned.
- If no valid instructions are identified, the function returns undefined.

matejcik, 2025-01-30:
Firmware-side, we already know how to identify and display the stake/unstake/withdraw instruction. However, if the transaction contains multiple instructions, the presence of a Stake instruction should not override the whole UI.
(consider a malicious transaction that unstakes your staking and in the same steps sends everything to an attacker)

we need to know the full structure of a stake/unstake/whatever transaction, with all the other instructions that are allowed to be there.

tomasklim, 2025-02-03:
**Overview:**

**Stake tx** https://solscan.io/tx/5Gx96KaWEWbkTKfMURFrxfM5qhkmypdJGeTHcQheQqcG463zfjNqp3B5riP6TUhjkJNCdGNbhBWGu681ZqPdYbZe
Instructions:
- Compute Budget: SetComputeUnitPrice
  - ComputeBudget111111111111111111111111111111
- System Program: createAccountWithSeed
  - 11111111111111111111111111111111
  - Trezor UI: create account, deposit
- Stake Program: initialize
  - Stake11111111111111111111111111111111111111
  - Trezor UI: initialize stake account, lockup authority
  - in data staker and withdrawer -> my address
- Stake Program: delegate
  - Stake11111111111111111111111111111111111111
  - Trezor UI: delegate from, vote account

_______________
**Unstake tx**
https://solscan.io/tx/5dbTNRryUSqcdcpvyJ2LKkBVyXMv9QacabyTbQqzDuURMLZ6Ex3tvY8qP4oEEYGze64CYguy8ThJoTeJDideB6jq
Instructions:
- Compute Budget: SetComputeUnitPrice
  - ComputeBudget111111111111111111111111111111
- Stake Program: deactivate
  - Stake11111111111111111111111111111111111111
  - Trezor UI: Deactivate stake account

_____________
**Claim tx**
https://solscan.io/tx/3umQeenCTcz8Q92kw5ehMDEAg2UKa5iPa39xQUQkdgJu7TLVPj5nw9wvwK1ksoC6QTYZ4NbPrXKJNjx2vqJuUedF
Instruction:
- Compute Budget: SetComputeUnitPrice
  - ComputeBudget111111111111111111111111111111
- Stake Program: withdraw
  - Stake11111111111111111111111111111111111111
  - TrezorUI: withdraw stake, from, to
    - should `to` be mine

vytick, 2025-02-07:
Imo the most important instructions in stake program are `Authorize` or its variants, where user sets withdraw authority and `Withdraw` where withdrawer specifies where the SOL is being sent after withdrawing from stake account.

fyi: there are also fresh [2 new instructions](https://github.com/solana-program/stake/blob/6d8ae65db1d5c4b85b508a7379fcd09d7bc24741/clients/js/src/generated/instructions/index.ts#L20C1-L21C29) in stake program: 

- moveLamports
- moveStake

Hannsek, 2025-02-13:
Designs for:
- [TS5](https://www.figma.com/design/n43caWJFGJlT167AQcxruu/branch/VBdd21hkXKtry1Ao2qU34h/%E3%80%94-Prod-%E3%80%95Safe-5?m=auto&node-id=10089-384&t=hiBmOyDQqcTqGVtC-1)
- TS3 (TBD)
- TT (TBD)

romanz, 2025-02-18:
Following [the work on TS5](https://github.com/trezor/trezor-firmware/pull/4639), I have a question about the fee-related layouts:

[image]

In #4641, I modified the code to calculate [base fee and priority fee](https://solana.com/docs/core/fees), but I am not sure how `Max rent fee` should be calculated.

Any suggestion will be appreciated :)

Hannsek, 2025-02-18:
Cc @tomasklim @vytick @matejcik

romanz, 2025-02-18:
Another two questions:

1. Would it be OK to show the vote account (in case it's not Everstake), e.g.:

[image]

2. Since I am currently using [the same UI flow as was used for ETH staking confirmation](https://github.com/trezor/trezor-firmware/blob/45adcd963a4f992b6545f6d52ad4a9412674db57/core/src/trezor/ui/layouts/delizia/__init__.py#L819), IIUC it doesn't support multiple info screens (to show the stake account and the blockhash):

[image]

[image]

which differs from Figma:

[image]

Would it be OK to show the stake account (as a single info screen), and the blockhash in the next confirmation screen?

[image]

[image]

romanz, 2025-02-18:
Also, just to be sure - which account should be checked in order to show the `"The current wallet isn't the SOL staking withdraw authority."` warning?

Is it the `"staker"` and/or `"withdrawer"` in the `"Stake Program: initialize"` instruction (e.g. see below)?

[image]

Hannsek, 2025-02-19:
> Would it be OK to show the vote account (in case it's not Everstake), e.g.:

If it won't be everstake, we definitely want to show the staking account (vote account). Just please use monofont if possible.

> Would it be OK to show the stake account (as a single info screen), and the blockhash in the next confirmation screen?

Then blockhash info would be visible in the main flow. We do not want that as it doesn't really give user much added value.
Can we maybe add more "items" into the "menu" component? Like there would be 4?

> Also, just to be sure - which account should be checked in order to show the "The current wallet isn't the SOL staking withdraw authority." warning?
> 
> Is it the "staker" and/or "withdrawer" in the "Stake Program: Initialize" instruction (e.g. see below)?

I believe it is a withdrawer. But I am not sure about that. Maybe @gabrielKerekes will help.

gabrielKerekes, 2025-02-19:
> I believe it is a withdrawer.

Yes. I haven't checked previous communication on this issue or the code but a similar warning should be shown for `staker`.

Hannsek, 2025-02-19:
@gabrielKerekes how does that work?

gabrielKerekes, 2025-02-19:
> how does that work?

I'm not sure what you mean. The `staker` can also be different from the stake account owner. I'm not exactly sure what an use case is, but it can. I guess I create an account and want you to manage who my account delegates to.

Hannsek, 2025-02-19:
So I stake my sol but I won't have access to it as the staker will be different from myself?

gabrielKerekes, 2025-02-19:
You don't stake it, you only create the account. 

However since we're probably talking only about "predefined transactions"  [where it seems](https://github.com/trezor/trezor-firmware/pull/4639/files#diff-1a8b9f5f98bccab56e3a1ff91ad97d28891a1bfb6ca847899f43834dafa2b136R244) the user creates a stake account and then delegates, then I guess the `staker` can't be different from the signer account in this scenario. And actually it's already handled in [the code](https://github.com/trezor/trezor-firmware/pull/4639/files#diff-1a8b9f5f98bccab56e3a1ff91ad97d28891a1bfb6ca847899f43834dafa2b136R259). Sorry for the confusion.

romanz, 2025-02-25:
There are some followups here: https://github.com/trezor/trezor-firmware/issues/4672

romanz, 2025-03-18:
https://github.com/trezor/trezor-firmware/issues/4672 is completed.

PR #4639 (2025-02-25): feat(core): add Solana staking confirmation dialog
Resolves #4560.

## Figma

- [TS5](https://www.figma.com/design/n43caWJFGJlT167AQcxruu/branch/VBdd21hkXKtry1Ao2qU34h/%E3%80%94-Prod-%E3%80%95Safe-5?node-id=10089-384&p=f&t=K4Z35CX0YcuJD0i7-0)
- [TS3](https://www.figma.com/design/ZEoQDp4EtHaAJHvYNxExHb/branch/Hq1fECKKChSRc7oj1RoxC0/%E3%80%94-Prod-%E3%80%95Safe-3?node-id=4627-2547&t=5ulivGq9P5NbqQBI-11)
- [Model T](https://www.figma.com/design/WbzQVEk1WzP0n9lRMJUB9S/branch/oWWyBfZ5XWedYGqzecCh07/%E3%80%94-Prod-%E3%80%95Model-T?node-id=2458-586&t=88U3GyLPYSZXjsW6-11)

## TODOs

- [x] show vote ac [... cut]
closed 2025-03-18 by romanz