trezor/trezor-firmware #4562: Feature Request: Chia (XCH) Integration in Trezor

Chiastream, 2025-01-30:
1. Coin Information

    Coin Name: Chia
    Ticker Symbol: XCH
    Blockchain Type: Custom Proof-of-Space-and-Time (PoST) blockchain
    Official Website: https://www.chia.net/
    Whitepaper: https://www.chia.net/greenpaper/
    Block Explorer: https://www.spacescan.io/

2. Why Should This Coin Be Supported?

    Innovative Technology: Chia uses Proof-of-Space-and-Time (PoST), which is an eco-friendly alternative to Proof-of-Work.
    Growing Adoption: Chia has an active developer community and increasing enterprise adoption.
    Security & Decentralization: Chia’s blockchain is designed for security and long-term sustainability, making it a great fit for cold storage solutions like Trezor.
    High User Demand: Many Chia holders seek secure hardware storage, and support from Trezor would provide an additional layer of security.
    Existing Hardware Wallet Support: Some competitors, like Ledger, have already announced partial support for XCH.

3. Technical Details

    GitHub Source Code: https://github.com/Chia-Network/chia-blockchain
    API Documentation: https://docs.chia.net/
    Current Wallet Support: Chia has its official wallet (Chia Wallet, Sage), but hardware wallet integration is still limited.
    Similar Coins Supported by Trezor: Other custom blockchain-based coins like Monero, Polkadot, and Solana are already supported.

4. Community Interest

    Chia has a strong and engaged community with 100k+ members across social media platforms.
    Discussions on Reddit, Discord, and the Chia Forum show increasing demand for Trezor support.
    Example discussions:
        [Chia Discord Community](https://chiaforum.com/)
        [Chia Forum](https://chiaforum.com/)
        [Chia Reddit Discussion](https://www.reddit.com/r/chia/)

Hannsek, 2025-01-31:
Hey, thanks for the proposal. We'll review it internally and will see if we, or someone else, would implement it.

Chiastream, 2025-02-06:
> Hey, thanks for the proposal. We'll review it internally and will see if we, or someone else, would implement it.

Thank you for considering the proposal. I appreciate your time and look forward to hearing your decision. If you have any questions, the Chia team is active and available on the Discord channel.