trezor/trezor-firmware #4567: Add PIVX PR closed and muted

Fuzzbawls, 2025-02-02:
A pull request to add additional support for PIIVX was closed on the basis of 'lack of user interest' (a rather weak basis). When comments started to flow, the ability to comment was then restricted. This is not how OSS development should happen.

PIVX submitted the work required, offered support, and even has the required infastructure to run nodes independently. PIVX is one of the only blockchains to offer alternate means (software) of managing user funds on hardware devices, and yet this is viewed as a negetive.

https://x.com/jskitteh/status/1886098835706912774
[image]

https://x.com/Fuzzbawls/status/1885847166478164035
[image]

Fuzzbawls, 2025-02-02:
The pull request in question: https://github.com/trezor/trezor-firmware/pull/3023

prusnak, 2025-02-02:
Please don't spam github.

closed 2025-02-02 by prusnak