trezor/trezor-firmware #4586: Bootloader Reproducibility issue in Trezor Firmware Build

keraliss, 2025-02-06:
Hi, this is keraliss from walletscrutiny. While testing firmware reproducability of the hardware wallets, mainly trezorSafe3, trezorSafe5 and trezorT, we hit a roadblock with the bootloader. Here's what we found:

- Build output shows bootloader hashes that don't match any downloadable bootloader
- We can't independently verify the bootloader's reproducibility
- It seems the bootloader is generated during the build process, but we have no way to confirm it matches the original

Can you help us understand:
- How the bootloader is generated
- How we can verify its reproducibility
- Why the bootloader hashes don't match any existing binaries

Thanks in advance!

Giszmo, 2025-02-06:
Do I understand you right? The firmware **contains** the bootloader binary of a prior revision of the firmware? So firmware 5 contains bootloader 3 and might get published together with a bootloader 5 that might get embedded in a later firmware? So we are accepting a binary blob cause it's checked in to the repo and we never checked the reproducibility of that bootloader blob? We need to adjust our scripts then ... So a failure to reproduce in above example bootloader 5 would not impede marking firmware 5 as reproducible as it contains bootloader 3, so if that is reproducible, we can treat firmware 5 as such, too.

matejcik, 2025-02-07:
That is exactly correct, yes.

I am closing the issue but feel free to continue discussion if you have more questions. Also may I point you to https://forum.trezor.io/ where I'm also responsive, which may be a better place for questions of this nature?

closed 2025-02-07 by matejcik

matejcik, 2025-02-07:
While I have you on the line, I'll point out that we switched to using internal model names even in the reproducible scripts and directory names.
Meaning that instead of `T` we now use `T2T1` and instead of `R` we use `T2B1`, both in arguments to build-docker.sh and in the names of the reproducible directories.

Giszmo, 2025-02-08:
I see https://github.com/trezor/trezor-firmware/tree/main/core/embed/models/T2B1/bootloaders was updated with the message "chore(core): add signed bootloader 2.1.8 for T2T1, T2B1". I searched briefly but can't find how these bin files were compiled.

matejcik, 2025-02-09:
again,

> bootloader is built from bootloader tags: `core/blX.Y.Z`, so you need to do the reproducible build from there.

specifically, the reproducible build is produced from [this commit](https://github.com/trezor/trezor-firmware/commit/656a86b8a0c501be380c3db8e3560039701f832c) tagged `core/bl2.1.8`
see arguments to `build-docker.sh` to know how to repro build a bootloader -- iirc you'll need to specify your models (`T,R` in that tag) and your targets (`bootloader`)