trezor/trezor-firmware #4657: Please add support for click >= 8.2.

sorenstoutner, 2025-02-21:
I am the maintainer of the Debian trezor-firmware package.

https://tracker.debian.org/pkg/python-trezor

The maintainers of the click package would like to upgrade to the upcoming 8.2 release.

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1098252

Would it be possible to update trezor-firmware to be compatible with click >= 8.2?

prusnak, 2025-02-23:
FTR 8.2.0 has not been released yet - here is the changelog: https://github.com/pallets/click/blob/main/CHANGES.rst#version-820

prusnak, 2025-02-23:
@sorenstoutner do you see any issues if you ease the dependency requirement and try to use python3-trezor with click==8.2.0 ?

matejcik, 2025-02-24:
reading the changelog, it appears that click 8.2 should not be breaking for us. but someone needs to try first.

sorenstoutner, 2025-02-24:
I maintain the Debian trezor package, but I do not personally own any trezor hardware to test it with (I ended up maintaining the trezor package, because it is a dependency of electrum, which I also maintain).  Given how serious some people's hardware wallets are to them, I thought it best to ask here before releasing a version that causes issues.

sorenstoutner, 2025-02-25:
I should mention that it builds fine against click 8.2.0 (the tag, which I assume is going to become the actual release).

https://github.com/pallets/click/pull/2844

This includes passing all the build checks.  But I don't know if any of those are really focused on checking for incompatibilities with new versions of click.

matejcik, 2025-02-25:
we have a tox config that builds and does some smoke-testing against different Click versions.
i'd accept a PR adding click 8.2 to the matrix and raising the requirement. then if any incompatibility appears outside that, we'll solve it as a bug

sorenstoutner, 2025-02-26:
Thanks.  I will reflect those changes in the current Debian package.  I really appreciate your assistance.

closed 2025-02-26 by sorenstoutner