trezor/trezor-firmware #4672: Followups for #4639

romanz, 2025-02-25:
> I observed some some deviations from the Figma deign.
> 
> * Delizia:
>   
>   * In the staking flow, the `Provider` has a different font.
>   * In all flows, the account info menu item has title  `Send from` instead of  `Account  info`
>   * In all flows, the confirm amount menu uses extra tap to cancel screen
> * Caesar:
>   
>   * In the claim flow, there should be a  title `Claim` and a question below.  The question is directly in the title.
> * Bolt:
>   
>   * In the first screen of the stake flow, the fonts `Stake SOL` and `Provider` are swapped.
>   * In all flows in Blockhash and Stake account menu items, there should be a bigger padding between the description and the value
> 
> They do not break the functionality so it is not critical to fix them all. If some of them are easy to fix, then let's do it.

https://github.com/trezor/trezor-firmware/pull/4639#issuecomment-2680968929

Hannsek, 2025-03-13:
@romanz Are you planning to work on this anytime soon?

romanz, 2025-03-17:
`In all flows, the confirm amount menu uses extra tap to cancel screen` marked as done - since it behaves similarly on BTC/ETH signing.

[image]
[image]

romanz, 2025-03-18:
`In all flows in Blockhash and Stake account menu items, there should be a bigger padding between the description and the value` marked as done - there is an inconsistency between `stake` and `unstake`/`claim` screens in Figma.

closed 2025-03-18 by romanz