trezor/trezor-firmware #4710: Not possible to confirm change of T3T1 brightness by taping

bosomt, 2025-03-02:
**Describe the bug**
Not possible to confirm change of T3T1 brightness by taping.
You have to swipe, tap to confirm message needs to be changed to swipe to confirm

**Steps to reproduce:**
1. a
2. b
3. c

**Info:**
 - Suite version: desktop 25.3.0 (21292fda4b99364182a872b4eefbc9986a6da622)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuiteDev/25.3.0 Chrome/132.0.6834.194 Electron/34.1.0 Safari/537.36
 - OS: MacIntel
 - Screen: 1470x956
 - Device: Trezor T3T1 2.8.9 regular (revision 7614a29c9a9cc138fff446937b7d9489fd57821f)
 - Transport: BridgeTransport 2.0.33

Hannsek, 2025-03-03:
The tap should be detected under the brightness bar.

bosomt, 2025-03-04:
**QA OK**

tap and swipe works as expected 🎉 

**Info:**
 - Suite version: desktop 25.3.0 (2a761d42998cb22ef837ae358cb9aaa39ea2123f)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuiteDev/25.3.0 Chrome/132.0.6834.194 Electron/34.1.0 Safari/537.36
 - OS: MacIntel
 - Screen: 1470x956
 - Device: Trezor T3T1 2.8.9 regular (revision fad9682201cf9289bba2adb66e6e07ed1cf78936)
 - Transport: BridgeTransport 2.0.33