trezor/trezor-firmware #4718: TROPIC01 model follow up

ibz, 2025-03-03:
This is a follow-up for #4172 to be done once the TROPIC01 model is open sourced.

* [ ] Revert commit that disables tropic in the emulator
* [ ] Start model as part of the Python unit tests runner (`core/tests/run_tests.sh`)

matejcik, 2025-03-10:
related (optiga side): https://github.com/trezor/trezor-firmware/pull/3335/commits/cde5a70c5c90c498756f635e07810c70625b3580

matejcik, 2025-04-01:
another item:

we should split the CI matrix, so that a build with `DISABLE_TROPIC=1` (without the model) is used for the integration test pipelines (device-tests, click-tests etc)
and a build with `DISABLE_TROPIC=0` (with the model) is used for core unit tests