trezor/trezor-firmware #4752: T3T1 missing last word of CS message when creating 20 word backup

bosomt, 2025-03-09:
**Describe the bug**

T3T1 missing last word of CS message when creating 20 w backup
`    "reset__words_written_down_template": "Mám zapsáno postupně všech {0} slov.",`

**Steps to reproduce:**
1. a
2. b
3. c

**Info:**
 - Suite version: desktop 25.3.1 (f789397f0b8634133dca90b1cee6e7574dbd1f84)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) TrezorSuite/25.3.1 Chrome/132.0.6834.194 Electron/34.1.0 Safari/537.36
 - OS: MacIntel
 - Screen: 1470x956
 - Device: Trezor T3T1 2.8.9 regular (revision fad9682201cf9289bba2adb66e6e07ed1cf78936)
 - Transport: BridgeTransport 2.0.33

**Screenshots:**

[image]

bosomt, 2025-03-10:
Same issue for `"fido__title_credential_details": "Podrobnosti přihlašovacích údajů",` when registering/approving FIDO2
i only see `Podrobnosti přihlašovacích...` on T3T1 display

prusnak, 2025-03-10:
Is there a way how to detect these overflows at compile time? I guess not.

If not, is there a way how to detect these overflows while running tests?

obrusvit, 2025-03-10:
The issue with the czech `"reset__words_written_down_template": "Mám zapsáno postupně všech {0} slov.",` can be fixed by using the newer version of TT Satoshi font.

obrusvit, 2025-03-12:
This of course doesn't solve our general issue of overflowing texts but the newer fonts fixed this particular case. https://github.com/trezor/trezor-firmware/pull/4773

[image]

closed 2025-03-24 by obrusvit