trezor/trezor-firmware #4817: TS5 doesn't react on ssh-keygen -t ecdsa-sk

elkimek, 2025-03-24:
**Describe the bug**
TS5 doesn't react when prompted with ```ssh-keygen -t ecdsa-sk``` but TT works just fine

**Firmware version and revision**
Latest

**Desktop/smartphone setup (please complete the following information):**
 - OS: pop_OS! latest

**To Reproduce**
Steps to reproduce the behavior:
1. $ ssh-keygen -t ecdsa-sk
2. Nothing happens on TS5

**Expected behavior**
TS5 should react with FIDO2 registration as shown in the docs https://trezor.io/learn/a/ssh-with-trezor

prusnak, 2025-04-16:
Do you use Universal firmware or Bitcoin-only one?

elkimek, 2025-04-17:
Universal on TT
Bitcoin-only on TS5

prusnak, 2025-04-17:
> Universal on TT
> Bitcoin-only on TS5

This explains it.

Bitcoin-only is not so surprisingly Bitcoin-only and does not have FIDO2 capabilities.
Install Universal firmware on TS5 and you should be good to go.