trezor/trezor-firmware #4820: Bytes overflow in latest firmware version

0xmichalis, 2025-03-24:
**Describe the bug**
Signing a quite large message is resulting in a "bytes overflow" error.

[image]

**Firmware version and revision**
v1.8.0

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Rabby
 - OS: Linux
 - Browser: Brave
 - Version 1.76.81

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://www.atomiclore.io/nftProfile/base/0xaf2430e1d91c7e22bb7d84588bccf8bc8301a09c/154
2. Click on `Sign`
3. Select `Comment` in the Provenance Templates
4. Paste below comment
```
blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah blah
``` 
5. Click `Continue` then `Sign`
6. See error

**Expected behavior**
I am able to sign any message regardless of its size.

**Additional context**
I was able to sign the message above (or at least a message of equivalent character count) just fine from a non-Trezor account via Rabby.

prusnak, 2025-03-24:
Firmware 1.8.0 is not the latest

Update to latest and report back

prusnak, 2025-03-24:
There is a high chance this is not possible with Trezor One and you need to get a newer version to sign longer messages.

matejcik, 2025-03-24:
Trezor One has a limit of 1024 bytes for message size.

Perhaps Trezor Connect should know this and return a nicer error message.

0xmichalis, 2025-03-24:
Yeah, if the limit is a deal breaker then at least a more helpful message would be welcome.

matejcik, 2025-03-25:
can we move the issue to Suite repo @Hannsek please?
cc @mroz22

mroz22, 2025-03-25:
yes. we can do it.