trezor/trezor-firmware #4940: T3T1 - Missing good to know screen when recovering multishare backup.

bosomt, 2025-04-22:
**Describe the bug**
T3T1 - Missing good to know screen when recovering multishare backup.
See current screens when recovering multishare backup

See discussion here: https://satoshilabs.slack.com/archives/C02BY51RD8R/p1744192881318409

**Steps to reproduce:**
1. a
2. b
3. c

**Info:**
 - Suite version: web 25.5.0 (b2c7a86fbc1035334e081b0c65e3daeabfcc573d)
 - Browser: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/135.0.0.0 Safari/537.36
 - OS: MacIntel
 - Screen: 1470x956
 - Device: Trezor T3T1 2.8.9 regular (revision fad9682201cf9289bba2adb66e6e07ed1cf78936)
 - Transport: BridgeTransport 2.0.33

**Screenshots:**

[image]
[image]
[image]
[image]

Hannsek, 2025-04-22:
It seems that this is not only on T3T1 but on all core devices.

Hannsek, 2025-04-22:
Figma file [here](https://www.figma.com/design/n43caWJFGJlT167AQcxruu/--Prod---Safe-5?node-id=3075-4024&t=gwNeLSt7J10Gsgld-4)