trezor/trezor-firmware #4967: ClusterFuzzLite PR fuzzing job takes >1h

romanz, 2025-04-28:
**Describe the bug**

[image]
https://github.com/trezor/trezor-firmware/actions/workflows/cflite.yml

Most of the time is spent downloading the corpus for the fuzzer:

[image]

IIUC, it seems that the corpus is not found - but the downloads fails only after waiting for >1h.

**Expected behavior**
I am not sure how large should be the corpus, but it probably shouldn't take more than a few minutes to download it.

**Additional context**
This feature was added in https://github.com/trezor/trezor-firmware/pull/2420.

@AdamKorcz could you please take a look?

mmilata, 2025-04-29:
Tried something but now it gets stuck in a different place (looks like uploading artifact instead of downloading it?) #4974

romanz, 2025-05-02:
https://github.com/trezor/trezor-firmware/actions/runs/14790752143/job/41527256536?pr=4988 failed after ~6 hours.

<details>

<summary> Traceback </summary>

```
2025-05-02 07:27:04,724 - root - INFO - Downloading corpus for fuzzer to /github/workspace/cifuzz-corpus/fuzzer.
Traceback (most recent call last):
  File "/opt/oss-fuzz/infra/cifuzz/run_fuzzers_entrypoint.py", line 97, in <module>
    sys.exit(main())
  File "/opt/oss-fuzz/infra/cifuzz/run_fuzzers_entrypoint.py", line 93, in main
    return run_fuzzers_entrypoint()
  File "/opt/oss-fuzz/infra/cifuzz/run_fuzzers_entrypoint.py", line 63, in run_fuzzers_entrypoint
    result = run_fuzzers.run_fuzzers(config)
  File "/opt/oss-fuzz/infra/cifuzz/run_fuzzers.py", line 316, in run_fuzzers
    if not fuzz_target_runner.run_fuzz_targets():
  File "/opt/oss-fuzz/infra/cifuzz/run_fuzzers.py", line 131, in run_fuzz_targets
    result = self.run_fuzz_target(target)
  File "/opt/oss-fuzz/infra/cifuzz/run_fuzzers.py", line [25](https://github.com/trezor/trezor-firmware/actions/runs/14790752143/job/41527256536?pr=4988#step:5:26)9, in run_fuzz_target
    return fuzz_target_obj.fuzz()
  File "/opt/oss-fuzz/infra/cifuzz/fuzz_target.py", line 175, in fuzz
    self._download_corpus()
  File "/opt/oss-fuzz/infra/cifuzz/fuzz_target.py", line 125, in _download_corpus
    self.clusterfuzz_deployment.download_corpus(self.target_name,
  File "/opt/oss-fuzz/infra/cifuzz/clusterfuzz_deployment.py", line 147, in download_corpus
    self.filestore.download_corpus(corpus_name, corpus_dir)
  File "/opt/oss-fuzz/infra/cifuzz/filestore/github_actions/__init__.py", line 105, in download_corpus
    return self._download_artifact(self.CORPUS_PREFIX + name, dst_directory)
  File "/opt/oss-fuzz/infra/cifuzz/filestore/github_actions/__init__.py", line 121, in _download_artifact
    if not self._raw_download_artifact(name, temp_dir):
  File "/opt/oss-fuzz/infra/cifuzz/filestore/github_actions/__init__.py", line 139, in _raw_download_artifact
    artifact = self._find_artifact(name)
  File "/opt/oss-fuzz/infra/cifuzz/filestore/github_actions/__init__.py", line 111, in _find_artifact
    artifact = github_api.find_artifact(name, artifacts)
  File "/opt/oss-fuzz/infra/cifuzz/filestore/github_actions/github_api.py", line 98, in find_artifact
    for artifact in artifacts:
  File "/opt/oss-fuzz/infra/cifuzz/filestore/github_actions/github_api.py", line 72, in _get_items
    response = _do_get_request(url, params=params, headers=headers)
  File "/opt/oss-fuzz/infra/retry.py", line 74, in _wrapper
    result = func(*args, **kwargs)
  File "/opt/oss-fuzz/infra/cifuzz/filestore/github_actions/github_api.py", line 57, in _do_get_request
    return requests.get(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/sessions.py", line 587, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/sessions.py", line 701, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/requests/adapters.py", line 489, in send
    resp = conn.urlopen(
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 716, in urlopen
    httplib_response = self._make_request(
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 468, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/usr/local/lib/python3.10/site-packages/urllib3/connectionpool.py", line 463, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/local/lib/python3.10/http/client.py", line 1375, in getresponse
    response.begin()
  File "/usr/local/lib/python3.10/http/client.py", line 318, in begin
    version, status, reason = self._read_status()
  File "/usr/local/lib/python3.10/http/client.py", line [27](https://github.com/trezor/trezor-firmware/actions/runs/14790752143/job/41527256536?pr=4988#step:5:28)9, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/usr/local/lib/python3.10/socket.py", line 705, in readinto
    return self._sock.recv_into(b)
  File "/usr/local/lib/python3.10/ssl.py", line 1[30](https://github.com/trezor/trezor-firmware/actions/runs/14790752143/job/41527256536?pr=4988#step:5:31)7, in recv_into
    return self.read(nbytes, buffer)
  File "/usr/local/lib/python3.10/ssl.py", line 1163, in read
    return self._sslobj.read(len, buffer)
KeyboardInterrupt
```

</details>

romanz, 2025-05-02:
It seems to be stuck downloading GitHub artifacts:
```
  File "/opt/oss-fuzz/infra/cifuzz/filestore/github_actions/__init__.py", line 139, in _raw_download_artifact
    artifact = self._find_artifact(name)
  File "/opt/oss-fuzz/infra/cifuzz/filestore/github_actions/__init__.py", line 111, in _find_artifact
    artifact = github_api.find_artifact(name, artifacts)
  File "/opt/oss-fuzz/infra/cifuzz/filestore/github_actions/github_api.py", line 98, in find_artifact
    for artifact in artifacts:
  File "/opt/oss-fuzz/infra/cifuzz/filestore/github_actions/github_api.py", line 72, in _get_items
    response = _do_get_request(url, params=params, headers=headers)
  File "/opt/oss-fuzz/infra/retry.py", line 74, in _wrapper
    result = func(*args, **kwargs)
```

romanz, 2025-05-02:
BTW, how many artifacts do we have here? 
Asking, since the code seems to list and iterate over all of them...
https://github.com/google/oss-fuzz/blob/c2ddf8318286bda8139773b55346bf0b06da2271/infra/cifuzz/filestore/github_actions/__init__.py#L149
https://github.com/google/oss-fuzz/blob/c2ddf8318286bda8139773b55346bf0b06da2271/infra/cifuzz/filestore/github_actions/github_api.py#L96

mmilata, 2025-05-02:
We produce over 200 artifacts per workflow run. I think all get deleted after 7 days or sooner. Guessing like 10-100k artifacts in the repo.

romanz, 2025-05-06:
IIUC, there is no way to filter for specific artifact(s):
https://docs.github.com/en/rest/actions/artifacts?apiVersion=2022-11-28#list-artifacts-for-a-repository
So I guess it's possible to use a separate repository to store the corpus data...

mmilata, 2025-05-06:
Looks like it can filter by `name`. But no need to do this if the corpus is never uploaded.

romanz, 2025-06-05:
Disabled artifacts download in https://github.com/trezor/trezor-firmware/pull/4998

closed 2025-06-05 by romanz