trezor/trezor-firmware #5039: Click test is flaky when invoked via `make test_emu_click_ui_multicore`

romanz, 2025-05-14:
Following https://github.com/trezor/trezor-firmware/pull/5028, there were several CI failures:

https://github.com/trezor/trezor-firmware/actions/runs/15014975444/job/42191021080?pr=4882
```
=========================== short test summary info ============================
FAILED ../tests/click_tests/test_repeated_backup.py::test_repeated_backup - trezorlib.exceptions.TrezorFailure: ActionCancelled: Cancelled
====== 1 failed, 51 passed, 17 skipped, 1 ui_failed in 104.57s (0:01:44) =======
```
https://github.com/trezor/trezor-firmware/actions/runs/15012996324/job/42185204774
```
=========================== short test summary info ============================
FAILED ../tests/click_tests/test_repeated_backup.py::test_repeated_backup - trezorlib.exceptions.TrezorFailure: ActionCancelled: Cancelled
====== 1 failed, 48 passed, 20 skipped, 1 ui_failed in 136.71s (0:02:16) =======
```
https://github.com/trezor/trezor-firmware/actions/runs/15016209550/job/42194974663
```
=========================== short test summary info ============================
FAILED ../tests/click_tests/test_repeated_backup.py::test_repeated_backup - trezorlib.exceptions.TrezorFailure: ActionCancelled: Cancelled
====== 1 failed, 56 passed, 12 skipped, 1 ui_failed in 120.98s (0:02:00) =======
```

romanz, 2025-05-14:
Reproduced with:
```
$ stress-ng --matrix 12 -t 1m &
$ make -C core test_emu_click TESTOPTS="--durations 0 -k test_repeated_backup"
```
Debug log:
```
18011702727 trezor.ui DEBUG UI task was stopped: <generator object '_handle_input_iface' at 0x7b085795d840>
18011703467 apps.debug DEBUG 1 receive: <DebugLinkGetState>
18011703686 trezor.wire.message_handler DEBUG received message contents:
DebugLinkGetState {
    return_empty_state: True
    wait_layout: 0
}
18011703897 trezor.wire.codec.codec_context DEBUG [WebUSB] write: DebugLinkState
18011704436 apps.debug DEBUG 1 receive: <DebugLinkGetState>
18011704679 trezor.wire.message_handler DEBUG received message contents:
DebugLinkGetState {
    return_empty_state: False
    wait_layout: 2
}
18011712511 trezor.loop DEBUG finish: <generator object '__iter__' at 0x7b085797f360>
18011719456 trezor.loop DEBUG finish: <generator object 'timer_task' at 0x7b0857981600>
18011719831 trezor.wire.codec.codec_context DEBUG [WebUSB] write: DebugLinkState
18011720284 trezor.loop DEBUG finish: <generator object 'timer_task' at 0x7b08579831c0>
18011725106 apps.debug DEBUG 1 receive: <DebugLinkGetState>
18011725302 trezor.wire.message_handler DEBUG received message contents:
DebugLinkGetState {
    return_empty_state: False
    wait_layout: 2
}
18011725706 trezor.wire.codec.codec_context DEBUG [WebUSB] write: DebugLinkState
18011731061 apps.debug DEBUG 1 receive: <DebugLinkDecision>
18011731177 trezor.wire.message_handler DEBUG received message contents:
DebugLinkDecision {
    x: 120
    input: None
    swipe: None
    hold_ms: None
    button: None
    physical_button: None
    y: 90
}
18011734067 trezor.ui DEBUG UI task was stopped: <generator object '_handle_input_iface' at 0x7b0857981e20>
18011734624 apps.debug DEBUG 1 receive: <DebugLinkGetState>
18011734683 trezor.wire.message_handler DEBUG received message contents:
DebugLinkGetState {
    return_empty_state: True
    wait_layout: 0
}
18011734744 trezor.wire.codec.codec_context DEBUG [WebUSB] write: DebugLinkState
18011739059 apps.debug DEBUG 1 receive: <DebugLinkGetState>
18011739317 trezor.wire.message_handler DEBUG received message contents:
DebugLinkGetState {
    return_empty_state: False
    wait_layout: 2
}
18011747128 trezor.ui.layouts.common INFO ButtonRequest sent: success_share_confirm
18011747329 trezor.ui.layouts.common INFO ButtonRequest acked: success_share_confirm
18011747496 trezor.loop DEBUG finish: <generator object '__iter__' at 0x7b08579a4200>
18011747695 trezor.loop DEBUG finish: <generator object 'timer_task' at 0x7b08579a5fe0>
18011748147 trezor.wire.codec.codec_context DEBUG [WebUSB] write: DebugLinkState
18011749796 apps.debug DEBUG 1 receive: <DebugLinkDecision>
18011750075 trezor.wire.message_handler DEBUG received message contents:
DebugLinkDecision {
    x: None
    y: None
    swipe: 0
    hold_ms: None
    button: None
    physical_button: None
    input: None
}
18011753477 trezor.ui DEBUG UI task was stopped: <generator object '_handle_input_iface' at 0x7b08579a6800>
18011757534 apps.debug DEBUG 1 receive: <DebugLinkGetState>
18011757756 trezor.wire.message_handler DEBUG received message contents:
DebugLinkGetState {
    return_empty_state: True
    wait_layout: 0
}
18011757957 trezor.wire.codec.codec_context DEBUG [WebUSB] write: DebugLinkState
18011758552 trezor.wire.message_handler INFO [WebUSB] received message: Initialize
18011762211 trezor.wire.message_handler DEBUG received message contents:
Initialize {
    derive_cardano: None
    session_id: b"v'\x02\x92]\xf8\xc9\\\x18\x9e\xba\xac\x97\x8a>\xcd\n'\x93\x1c\xe1\xeao!\xcc\x86\xacv\xbb\x13\xd3*"
}
18011762400 trezor.loop DEBUG spawn new task: <generator object 'with_context' at 0x7b08579bc820>
18011762545 trezor.workflow DEBUG start: <generator object 'with_context' at 0x7b08579bc820>
18011762747 trezor.loop DEBUG close spawned task: <generator object 'with_context' at 0x7b08579bc820>
18011762832 trezor.workflow DEBUG close: <generator object 'with_context' at 0x7b08579bc820>
18011762913 trezor.workflow DEBUG default already started
18011765454 trezor.ui.layouts.common INFO ButtonRequest sent: success_backup
18011765574 trezor.ui.layouts.common INFO ButtonRequest acked: success_backup
18011765712 trezor.wire.message_handler DEBUG [WebUSB] cancelled: loop task was closed
18011765825 trezor.wire.codec.codec_context DEBUG [WebUSB] write: Failure
18011783275 trezor.loop DEBUG finish: <generator object 'timer_task' at 0x7b08579bfe20>
```

romanz, 2025-05-14:
Comparing with a successful run:

[image]

IIUC, if `Initialize` handler is scheduled **before** the last "fake" ButtonRequest is handled, it will be cancelled by the `workflow.close_others()` call here:
https://github.com/trezor/trezor-firmware/blob/12269d48148f69ee8563d4b477d98cc86e242c42/core/src/trezor/ui/layouts/common.py#L23-L36

matejcik, 2025-05-14:
right, because in this testcase the debuglink behavior is wrong:
https://github.com/trezor/trezor-firmware/blob/mmilata/homescreen-stream/python/src/trezorlib/debuglink.py#L662-L663

we're doing a synchronization event but with `return_empty_state=True`, instead of what the original code was doing:
https://github.com/trezor/trezor-firmware/blame/85bbc89eed05d13d3c3db005971c65743a4f9919/python/src/trezorlib/debuglink.py#L647-L674

from https://github.com/trezor/trezor-firmware/pull/4568

i'm wondering if we could keep `return_empty_state` but add back the correct wait mode?

matejcik, 2025-05-14:
> i'm wondering if we could keep `return_empty_state` but add back the correct wait mode?

(to do that we'd also have to modify the debuglink responder to handle `return_empty_state` + wait mode correctly)

PR #5041 (2025-05-19): test(core): explicitly confirm backup success
Fixes #5039.
closed 2025-05-19 by romanz